yamcs/yamcs #139: Data definition debugging

bitblit11, 2015-09-28:
Is there a way to debug telemetry and command definitions?  I got the baseline install with simulation working perfectly.  My own project uses the CCSDS protocol.  My client connects successfully.  I updated the xls files in the mdb directory to include my messages but the server raw data count is not incrementing.  The logs don't show anything helpful, as far as I can see.  Is there a way to see why pushed data was not parsed?  Or at least dropped message counts?

xpromache, 2015-09-28:
Hello,

Sorry, I do not understand what you are trying to achieve.
Which client connects successfully?
Did you replace our simulator with your own version?

bitblit11, 2015-09-28:
I replaced the simulator with my own data source and wrote .xls files in the mdb directory reflecting my telemetry and command definitions.   My source is also CCSDS but does not have the packet ID.  

yamcs-monitor shows my data source is connected.

xpromache, 2015-09-28:
One thing to know is that the TcpPacketProvider does not look in the xls file to split the incoming TCP stream to packets. 
There is a stream definition in XTCE but not implemented in Yamcs.

Instead, the CCSDS packets size is determined from the bytes 5+6. That should be the remaining length-1. If the data count is not increasing, it must be something wrong with this length.

I suggest you stop the yamcs server and do something like netcat your-data-source-ip your-data-source-port  >/tmp/packetdump.tmp

That should create a dump of the CCSDS packets in a file.

Then you can check that file with a hexdump (xxd) and even try to load it in standalone Yamcs packet viewer (bin/packet-viewer.sh, File->Open (or thereabouts) )

bitblit11, 2015-09-30:
I walked through the installation and simulation instructions with no problem.  There's not a lot of documentation on interfacing YAMCS to a real client.  Is there a YAMCS mailing list and/or forum?  Maybe an example of a real (non-simulated) client interfacing to a remote node via UDP CCSDS messages?

xpromache, 2015-10-01:
Hello,

Yes, I know there is not a lot of documentation. We (Space Applications Services) are improving things little by little as time allows.
Yamcs is being used in some operational centers related to ISS operations, however the documentation produced for them cannot be published. It would not be immediately applicable anyway.

Related to mailing list: I have opened once a group here: https://groups.google.com/forum/#!forum/yamcs 
but no message has been ever published there. Be welcome to publish the first message :)

As far as I understand, your client is the Data Source replacing our simulator, correct? 
But what do you mean with "real (non-simulated) client"? 
We usually call clients the applications at the other end of the chain - that visualize the telemetry or send commands.

Maybe you can describe better your scenario.

nicolae

bitblit11, 2015-10-02:
I posted to the forum.  Closing this ticket.  Thanks.

closed 2015-10-02 by bitblit11