yamcs/yamcs #192: Install problem on ubuntu 14.04

shhan, 2016-07-19:
I downloaded yamcs from github.
and executed make-rpm.sh in the yamcs source directory.

My env. is Ubuntu 14.04.

Command is 
$> sudo ./make-rpm.sh

please check end side of message.

>>>>>>>>>>>>>>>>>>>>>>>>>>>>> Output Message <<<<<<<<<<<<<<<<<<<<<<<<<<<<
......
....
gulp-watch@4.3.8 node_modules/gulp-watch
├── path-is-absolute@1.0.0
├── object-assign@4.1.0
├── vinyl@0.5.3 (clone-stats@0.0.1, clone@1.0.2, replace-ext@0.0.1)
├── readable-stream@2.1.4 (buffer-shims@1.0.0, process-nextick-args@1.0.7, util-deprecate@1.0.2, string_decoder@0.10.31, inherits@2.0.1, core-util-is@1.0.2, isarray@1.0.0)
├── glob-parent@2.0.0 (is-glob@2.0.1)
├── chokidar@1.6.0 (inherits@2.0.1, async-each@1.0.0, is-binary-path@1.0.1, is-glob@2.0.1, readdirp@2.1.0)
├── anymatch@1.3.0 (arrify@1.0.1, micromatch@2.3.11)
├── vinyl-file@1.3.0 (graceful-fs@4.1.4, strip-bom@2.0.0, strip-bom-stream@1.0.0, vinyl@1.1.1)
└── gulp-util@3.0.7 (array-differ@1.0.0, beeper@1.1.0, lodash._reescape@3.0.0, lodash._reevaluate@3.0.0, lodash._reinterpolate@3.0.0, object-assign@3.0.0, array-uniq@1.0.3, replace-ext@0.0.1, fancy-log@1.2.0, minimist@1.2.0, has-gulplog@0.1.0, gulplog@1.0.0, lodash.template@3.6.2, through2@2.0.1, chalk@1.1.3, multipipe@0.1.2, dateformat@1.0.12)

gulp-filter@3.0.1 node_modules/gulp-filter
├── streamfilter@1.0.5 (readable-stream@2.1.4)
├── multimatch@2.1.0 (arrify@1.0.1, array-differ@1.0.0, array-union@1.0.2, minimatch@3.0.2)
└── gulp-util@3.0.7 (array-differ@1.0.0, beeper@1.1.0, object-assign@3.0.0, lodash._reescape@3.0.0, lodash._reevaluate@3.0.0, lodash._reinterpolate@3.0.0, replace-ext@0.0.1, array-uniq@1.0.3, minimist@1.2.0, fancy-log@1.2.0, has-gulplog@0.1.0, vinyl@0.5.3, gulplog@1.0.0, lodash.template@3.6.2, chalk@1.1.3, multipipe@0.1.2, through2@2.0.1, dateformat@1.0.12)

gulp-uglify@1.5.4 node_modules/gulp-uglify
├── uglify-save-license@0.4.1
├── deap@1.0.0
├── isobject@2.1.0 (isarray@1.0.0)
├── fancy-log@1.2.0 (time-stamp@1.0.1, chalk@1.1.3)
├── through2@2.0.1 (xtend@4.0.1, readable-stream@2.0.6)
├── uglify-js@2.6.4 (uglify-to-browserify@1.0.2, async@0.2.10, yargs@3.10.0, source-map@0.5.6)
├── vinyl-sourcemaps-apply@0.2.1 (source-map@0.5.6)
└── gulp-util@3.0.7 (array-differ@1.0.0, beeper@1.1.0, lodash._reescape@3.0.0, lodash._reevaluate@3.0.0, lodash._reinterpolate@3.0.0, array-uniq@1.0.3, object-assign@3.0.0, chalk@1.1.3, replace-ext@0.0.1, vinyl@0.5.3, lodash.template@3.6.2, has-gulplog@0.1.0, gulplog@1.0.0, minimist@1.2.0, multipipe@0.1.2, dateformat@1.0.12)

gulp-less@3.1.0 node_modules/gulp-less
├── object-assign@4.1.0
├── through2@2.0.1 (xtend@4.0.1, readable-stream@2.0.6)
├── vinyl-sourcemaps-apply@0.2.1 (source-map@0.5.6)
├── accord@0.23.0 (indx@0.2.3, semver@5.3.0, convert-source-map@1.2.0, fobject@0.0.4, resolve@1.1.7, glob@7.0.3, when@3.7.7, uglify-js@2.7.0, lodash@4.11.2)
├── gulp-util@3.0.7 (array-differ@1.0.0, beeper@1.1.0, lodash._reescape@3.0.0, lodash._reinterpolate@3.0.0, lodash._reevaluate@3.0.0, object-assign@3.0.0, array-uniq@1.0.3, replace-ext@0.0.1, fancy-log@1.2.0, has-gulplog@0.1.0, gulplog@1.0.0, minimist@1.2.0, lodash.template@3.6.2, chalk@1.1.3, vinyl@0.5.3, multipipe@0.1.2, dateformat@1.0.12)
└── less@2.7.1 (graceful-fs@4.1.4, mime@1.3.4, image-size@0.5.0, source-map@0.5.6, mkdirp@0.5.1, promise@7.1.1, errno@0.1.4)
+ gulp
/var/tmp/rpm-tmp.MCVmcM: 29: /var/tmp/rpm-tmp.MCVmcM: gulp: not found
+ mvn -e clean compile package -Dmaven.test.skip=true -Dmaven.buildNumber.doUpdate=false
[INFO] Error stacktraces are turned on.
[INFO] Scanning for projects...
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 0.073s
[INFO] Finished at: Tue Jul 19 15:55:45 KST 2016
[INFO] Final Memory: 4M/119M
[INFO] ------------------------------------------------------------------------
**[ERROR] The goal you specified requires a project to execute but there is no POM in this directory (/home/scidb/rpmbuild/BUILD/yamcs-0.29.5+r1c64291/yamcs-web). Please verify you invoked Maven from the correct directory. -> [Help 1]**
org.apache.maven.lifecycle.MissingProjectException: The goal you specified requires a project to execute but there is no POM in this directory (/home/scidb/rpmbuild/BUILD/yamcs-0.29.5+r1c64291/yamcs-web). Please verify you invoked Maven from the correct directory.
	at org.apache.maven.lifecycle.internal.LifecycleStarter.execute(LifecycleStarter.java:89)
	at org.apache.maven.DefaultMaven.doExecute(DefaultMaven.java:320)
	at org.apache.maven.DefaultMaven.execute(DefaultMaven.java:156)
	at org.apache.maven.cli.MavenCli.execute(MavenCli.java:537)
	at org.apache.maven.cli.MavenCli.doMain(MavenCli.java:196)
	at org.apache.maven.cli.MavenCli.main(MavenCli.java:141)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:57)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:606)
	at org.codehaus.plexus.classworlds.launcher.Launcher.launchEnhanced(Launcher.java:289)
	at org.codehaus.plexus.classworlds.launcher.Launcher.launch(Launcher.java:229)
	at org.codehaus.plexus.classworlds.launcher.Launcher.mainWithExitCode(Launcher.java:415)
	at org.codehaus.plexus.classworlds.launcher.Launcher.main(Launcher.java:356)
[ERROR] 
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MissingProjectException
error: Bad exit status from /var/tmp/rpm-tmp.MCVmcM (%build)

RPM build errors:
    bogus date in %changelog: Thu Nov 20 2010 nm
    bogus date in %changelog: Fri Jul 26 2007 nm
    Bad exit status from /var/tmp/rpm-tmp.MCVmcM (%build)
.....

xpromache, 2016-07-19:
(you don't need to sudo to makr-rpm, should normally work as a regular user)

RPM fails because it doesn't find the gulp command.

gulp is required for compiling the javascript web interface. You can either install gulp and npm (search for "gulp install ubuntu 14.04" ) or run make-rpm.sh -noweb which doesn't build the web interface (not sure what you want to use yamcs for if you need that or not).

shhan, 2016-07-21:
Thanks, xpromache.

I installed gulp and run the script as a regular user.
And I got success for one step, but Next I got the other problem.

For solving following things, I checked my javac version.
$> javac -version
javac 1.8.0_91
And, I removed the repository and clone new one.
And, I removed the ~/.m2 directory.  and something more.

but, I don't know what's the problem is.

scidb@scidb-h9-1170kr:~/work/yamcs$ mvn clean compile package -Dmaven.test.skip=true -Dmaven.buildNumber.doUpdate=false
[INFO] Scanning for projects...
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Build Order:
[INFO] 
[INFO] Yamcs
[INFO] Yamcs API
[INFO] Yamcs XTCE
[INFO] Yamcs Core
[INFO] Yamcs Simulation
[INFO]                                                                         
[INFO] ------------------------------------------------------------------------
[INFO] Building Yamcs 0.29.5
[INFO] ------------------------------------------------------------------------
[INFO] 
[INFO] --- maven-clean-plugin:2.5:clean (default-clean) @ yamcs ---
[INFO] 
[INFO] --- maven-source-plugin:2.2.1:jar-no-fork (attach-sources) @ yamcs ---
[INFO]                                                                         
[INFO] ------------------------------------------------------------------------
[INFO] Building Yamcs API 0.29.5
[INFO] ------------------------------------------------------------------------
[INFO] 
[INFO] --- maven-clean-plugin:2.5:clean (default-clean) @ yamcs-api ---
[INFO] Deleting /home/scidb/work/yamcs/yamcs-api/target
[INFO] 
[INFO] --- maven-resources-plugin:2.3:resources (default-resources) @ yamcs-api ---
[INFO] Using 'UTF-8' encoding to copy filtered resources.
[INFO] Copying 1 resource
[INFO] 
[INFO] --- maven-compiler-plugin:2.3.2:compile (default-compile) @ yamcs-api ---
[INFO] Compiling 64 source files to /home/scidb/work/yamcs/yamcs-api/target/classes
[INFO] -------------------------------------------------------------
[ERROR] COMPILATION ERROR : 
[INFO] -------------------------------------------------------------
**[ERROR] Failure executing javac, but could not parse the error:
javac: invalid target release: 1.8**
Usage: javac <options> <source files>
use -help for a list of possible options

[INFO] 1 error
[INFO] -------------------------------------------------------------
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary:
[INFO] 
[INFO] Yamcs ............................................. SUCCESS [0.430s]
[INFO] Yamcs API ......................................... FAILURE [0.308s]
[INFO] Yamcs XTCE ........................................ SKIPPED
[INFO] Yamcs Core ........................................ SKIPPED
[INFO] Yamcs Simulation .................................. SKIPPED
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 0.887s
[INFO] Finished at: Fri Jul 22 05:54:29 KST 2016
[INFO] Final Memory: 8M/150M
[INFO] ------------------------------------------------------------------------
[**ERROR] Failed to execute goal org.apache.maven.plugins:maven-compiler-plugin:2.3.2:compile (default-compile) on project yamcs-api: Compilation failure
[ERROR] Failure executing javac, but could not parse the error:
[ERROR] javac: invalid target release: 1.8
[ERROR] Usage: javac <options> <source files>**
[ERROR] use -help for a list of possible options
[ERROR] -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoFailureException
[ERROR] 
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <goals> -rf :yamcs-api

xpromache, 2016-07-22:
It's weird that  you get this error which normally means you don't have java 1.8.
Maybe you have two java versions and the other one gets picked up somehow by maven.

Can you try to run this to see what is says:
update-alternatives --config javac

fqqb, 2016-07-22:
And verify whether JAVA_HOME environment variable is not pointing to an older jdk.
`echo $JAVA_HOME`

shhan, 2016-07-24:
Thank you for your help. I at last build it successfully.

For this, 
I knew we need to check javac version and JAVA_HOME environment.

$> sudo update-alternatives --config javac
$> echo $JAVA_HOME
You may know whether it's correct or not,

For changing $JAVA_HOME in ubuntu 
$> sudo vi /etc/environment
and add 
JAVA_HOME="/usr/lib/...."
$> source /etc/environment
$> echo $JAVA_HOME
/usr/lib/jvm/java-8-openjdk-amd64/

Anyway, I really thank you for your help.

closed 2016-07-24 by shhan