yamcs/yamcs #194: RPM package install problem on Ubuntu 14.04

shhan, 2016-07-24:
I got the RPM file from make-rpm.sh script on yamcs package,
but failed to install by rpm command.

may I use alien command on Ubuntu.

$> cd ~/rpmbuild/RPMS/noarch
$> ls
yamcs-0.29.5+rdd31aad-10.noarch.rpm
$> rpm -U yamcs-0.29.5+rdd31aad-10.noarch.rpm 
rpm: RPM should not be used directly install RPM packages, use Alien instead!
rpm: However assuming you know what you are doing...
error: Failed dependencies:
	/bin/bash is needed by yamcs-0.29.5+rdd31aad-10.noarch
	/bin/sh is needed by yamcs-0.29.5+rdd31aad-10.noarch
	lsb is needed by yamcs-0.29.5+rdd31aad-10.noarch

xpromache, 2016-07-25:
Ubuntu uses deb packages not rpms.

So please convert the rpm to a deb with this command:
sudo alien --scripts yamcs-0.29.5+rdd31aad-10.noarch.rpm 

(you may need to install the alien tool for that: sudo apt-get install alien)
That should create a .deb file which you can install wiht dpkg -i.

shhan, 2016-07-25:
Thanks a lot, xpromache.

This is my little contribution for this issue.

For installing yamcs, you use the rpm command with rpm file in some system, but ubuntu usually use deb file for installing, then you have to change rpm to deb by alien command.

if you don't have alien,
$> sudo apt-get install alien

change rpm to deb
$> sudo alien --scripts yamcs-0.29.5+rdd31aad-10.noarch.rpm
install deb package
$> sudo dpkg -i yamcs_0.29.5+rdd31aad-11_all.deb

Then, you may see the installed yamcs package

$> ls /opt/yamcs/
bin  cache  etc  lib  log  mdb  web

closed 2016-07-25 by shhan