yamcs/yamcs #197: GET /api/links/:instance/:name returns 404

temecom, 2016-11-08:
When retrieving a specific link that is listed in the /api/links/:instance call the /api/links/:instance/:name call returns 404 Not Found. eg (with leo_spacecraft docker image) 

**http://localhost:8090/api/links/simulator**
Returns: 
{
  "link" : [ {
    "instance" : "simulator",
    "name" : "tm_realtime",
    "type" : "TcpTmProvider",
    "spec" : "local",
    "stream" : "tm_realtime",
    "disabled" : false,
    "status" : "OK",
    "dataCount" : 1228,
    "detailedStatus" : "OK, connected to simulator:10015, received 1228 packets"
  }, {
    "instance" : "simulator",
    "name" : "tm_dump",
    "type" : "TcpTmProvider",
    "spec" : "localDump",
    "stream" : "tm_dump",
    "disabled" : false,
    "status" : "OK",
    "dataCount" : 0,
    "detailedStatus" : "OK, connected to simulator:10115, received 0 packets"
  }, {
    "instance" : "simulator",
    "name" : "pp1",
    "type" : "SimulationPpProvider",
    "spec" : "{simulationDataPath=etc/gs-simulation.xml}",
    "stream" : "pp_realtime",
    "disabled" : false,
    "status" : "OK",
    "dataCount" : 3,
    "detailedStatus" : "OK"
  }, {
    "instance" : "simulator",
    "name" : "tc1",
    "type" : "TcpTcUplinker",
    "spec" : "local",
    "stream" : "tc_realtime",
    "disabled" : false,
    "status" : "OK",
    "dataCount" : 0,
    "detailedStatus" : "OK, connected to simulator:10025"
  } ]
}

**http://localhost:8090/api/links/simulator/tm_realtime**

Returns: 

404 Not Found

Tried various combinations but none work.

xpromache, 2016-11-08:
Hello, thanks for the report. I see in the [https://github.com/yamcs/yamcs/blob/master/yamcs-core/src/main/java/org/yamcs/web/rest/LinkRestHandler.java](sourcecode) that the URL is
http://localhost:8090/api/links/simulator/link/tm_realtime

It looks incorrect, but I will check with the person that implemented it if there is any reason.

fqqb, 2016-11-09:
The current version of the demo docker examples uses Yamcs 0.29.3. 
Your reported bug was addressed a while back in 9737ed7, but it did not make it in a release yet.

The current workaround like @xpromache mentioned is to insert the 'link' word as a path segment between the instance and the stream name. I have for now updated the online documentation: 

http://www.yamcs.org/docs/api/Get_Link/
http://www.yamcs.org/docs/api/Edit_Link/

In our next releases, we will support both with and without the 'link' word for a while.

fqqb, 2018-05-07:
> In our next releases, we will support both with and without the 'link' word for a while.

This has been in effect for a while already. Closing this old issue.

closed 2018-05-07 by fqqb