yamcs/yamcs #208: Total number of events in the archive

ampeixoto, 2017-06-13:
I'm running the Docker demo "A 5Hz Launch & Landing simulator" and querying for events with a link like this:
[http://localhost:8090/api/archive/simulator/events?pos=0&limit=10](http://localhost:8090/api/archive/simulator/events?pos=0&limit=10)

I will have to do pagination, so I need to the total number of events or at least the indication if there is a next page.

Is there a way to get this information?

fqqb, 2017-06-14:
To know if there's a next page, you could set your limit to 11, and only show 10.

If you have fast incoming events, using pos may not be sufficiently reliable either. You could alternatively check the date of the last fetched event and filter your next page with that as a stop time (if you're iterating in reverse chronologic order).

Total number of events is not something we have readily available as of yet. Theoretically you could download the event index records ( http://www.yamcs.org/docs/api/Download_Event_Index/ ) and add all the numbers up, but it's not so practical.

xpromache, 2017-06-14:
I see for example what sites that provide log management (such as logentries) are doing, is displaying a sort of timeline or histogram (basically showing the number of logs in a time interval) on top and then the logs on bottom:
https://blog.logentries.com/2014/01/5-tricks-to-become-a-logentries-power-user/

You could use the index to implement such timeline and at the same time to count the events. We should have this kind o functionality builtin the yamcs web...

ampeixoto, 2017-06-22:
Thanks for the suggestions.
Will try them when possible.

closed 2017-07-20 by ampeixoto