yamcs/yamcs #276: Yamcs web interface: MDB/Parameters/Chart - timestamp missing when hovering over chart

kstruyven, 2018-09-10:
When hovering over a chart of a MDB parameter, it only shows the value but not the timestamp. This makes it impossible to get a value at a specific time.

[image]

fqqb, 2018-09-11:
Thanks for the issue. I'll work on this. Probably will be via a timestamp indicator displayed on the x-axis that moves together with the vertical mouse locator.

fqqb, 2018-09-13:
Implemented in #279 Example:

[image]

fqqb, 2018-09-24:
Merged on master. This will be in the next 4.x release.

closed 2018-09-24 by fqqb