yamcs/yamcs #285: Yamcs API - archived parameter values 'limit' option ignored for high number

kstruyven, 2018-10-03:
When fetching archived parameter values from the Yamcs API, the 'limit' option in the request should indicate how much values are returned maximum. When setting a high limit (10000), I never get more then about 3200 values returned. It seems to be the max amount of data that can be returned. 
E.g. of URL that returns (only) 3200 values.
/archive/fsl-ops/parameters/FSL_EVENTS/Message?&start=2018-10-02T21:09:54.000Z&stop=2018-10-04T00:09:54.866Z&order=asc&limit=10000'

xpromache, 2018-10-04:
Maybe there is a bug in the usage of the stop (it somehow causes the read to end too early), can you try  providing a stop some hours in the future?

fqqb, 2020-09-09:
Closing because old and missing further input

closed 2020-09-09 by fqqb