yamcs/yamcs #294: docker-compose + nginx + location + proxy_pass + rewrite

vinnitu, 2018-11-15:
I need get access to yamcs server through nginx subpath location /yamcs

I use 

```
    location /yamcs {
        proxy_pass         http://localhost:8090;
        rewrite            ^/yamcs/(.*)  /$1 break;
    }

```
What I forget?

fqqb, 2018-11-16:
Would need a bit more information to be able to judge your use case correctly.
Is your purpose to access the Yamcs website via this proxy?  Are nginx and yamcs in two separate containers?

If so, then maybe first check if you receive html when you `curl http://localhost:8090` on the container that runs Yamcs. If that works then try the same on the container that runs nginx.

vinnitu, 2018-11-16:
I want to serve service by http://example.com/yamcs
In simple case we can add <base href="/yamcs/"> in html-head in our single page app, but need to change other links like src="/static..." to src="static..."

fqqb, 2018-11-19:
If you're building from source you could specify `--base-href` to the ng command (see build script in yamsc-web/packages/app/package.json). To make this dynamically configurable, is something I can take a look at.

fqqb, 2018-11-19:
And probably --deploy-url is required too.

closed 2018-11-22 by vinnitu