yamcs/yamcs #295: yamcsadmin parchive import

vinnitu, 2018-11-22:
We have historical telemerty date and wanna import it to yamcs parameter archive (to rocksdb).
What is the proper way to do it?

xpromache, 2018-11-23:
What format is your data?

Currently the only way to import some data is if it has been exported from
Yamcs or if it's a known packet format like CCSDS but we might consider
implementing some additional means if you tell me a bit about your usage
scenario.

nicolae

vinnitu, 2018-11-26:
csv - time,metric1,metric2,...

xpromache, 2018-11-27:
Does your CSV file include a measurement for all metrics? or you have some empty positions?

In any case, currently it's not easy to import such a file, but we hopefully will implement something in the following months.

Just for clarification - currently Yamcs needs a so called Mission Database (MDB) where all the parameters (metrics) are defined.
However, the parameter archive (based on rocksdb) can work in principle without it, we need a little work to make it more accessible from outside.

I suggest to keep this issue open, we will update it when we have something.