yamcs/yamcs #299: yamcs-web and archive tags

vinnitu, 2018-12-19:
Do you have any plans to append support archive tags in web client?

fqqb, 2019-01-10:
I can have a look at this once #305 is resolved

vinnitu, 2019-01-10:
[image]

We trying to implement tags functionality, do you interested to get it? Maybe you have more competent vision about it?

fqqb, 2019-01-10:
That's great, I'll not touch this then. 

I'm curious why a different page, I would argue it's useful to show tags together with the packet ranges, but all in one single band. This is what we do in Yamcs Studio (orange boxes are tags):

[image]

The timeline widget supports wrapping boxes in the same band (option `wrap: true`).

vinnitu, 2019-01-10:
[image]

version without wrapping

fqqb, 2021-04-20:
Closing as we no longer plan to support edit of tags on yamcs-web. Only read-only is implemented for legacy reasons.

Instead we are working towards a more feature-full timeline solution. This work is currently on separate branch (server parts only): https://github.com/yamcs/yamcs/tree/timeline

closed 2021-04-20 by fqqb