yamcs/yamcs #302: few instances with same xls config

vinnitu, 2019-01-03:
Hello!

Can we use the same xls config for multiple instances passing instance name (General tab) from yaml?

xpromache, 2019-01-03:
Not sure what you want to accomplish: in the general tab of the xls you
define the so called space system name.
For example in the refmdb spreadsheet you have REFMDB so all your
parameters will be of the shape /REFMDB/para_name. In that spreadsheet you
have in fact two space systems, so you also get parameters of the shape
/REFMDB/SUBSYS1/para_name

"instance" usually refers to a Yamcs instance - is like separate
monitoring& control domains if you want. Each instance has a name but that
name is not necessarily related to the space system names inside the MDB
(e.g. xls).

You can have multiple Yamcs instances referring to the same xls file, just
use the same mdb config in the two yamcs.instance1.yaml and
yamcs.instance2.yaml.
The MDB  inside one Yamcs Instance  is completely separate from another
instance.

nicolae

vinnitu, 2019-01-03:
Oook.. Thank you.

closed 2019-01-03 by vinnitu