yamcs/yamcs #303: multiple datalinks for one data source

vinnitu, 2019-01-03:
Hello again!

We have one data source (It means the same ccsds-apids). And it has two radio transmitters (datalinks). The second one is used for fault tolerance. And we want to see and compare data from both. How to handle such situation?

xpromache, 2019-01-03:
What if you make a second instance and configure the second link in the
second instance only? To create a second instance, just copy your
yamcs.<instance>.yaml file into yamcs.<secondinstance>.yaml and add
<secondinstance> in yamcs.yaml.

How do you monitor the data?

You could also make a second realtime processor in the same instance but we
never tested this configuration. To do that you can copy this:
 -  - class: org.yamcs.ProcessorCreatorService
     args:
        name: "realtime"
        type: "realtime"

in the same yamcs.<instance>.yaml and give another name "e.g. realtime1".

Maybe give it a try and let us know what happens.

xpromache, 2019-01-04:
Actually I realized is not that simple: in order to have two realtime
processors, you need to:
 - make another tm_stream (in yamcs.<instance>.yaml in the streamConfig
section
streamConfig:
 tm:
   - name: "tm_realtime1"
   - name: "tm_realtime"
...

- change one of your data links to put the data into the new stream
dataLinks:
 - name: tm_realtime1
   class: org.yamcs.tctm.TcpTmDataLink
   args:
     stream: tm_realtime1
     host: localhost
 ....

- make also another processor type in processor.yaml  (just copy the
realtime into realtime1) and change which stream data is read from:
realtime1:
 services:
    - class: org.yamcs.StreamTmPacketProvider
     args:
       streams: ["tm_realtime1"]
....

 so one processor reads data from one stream and the other processor from
the other stream.

- and finally the thing mentioned earlier in yamcs.<instance>.yaml to start
the processor with the new name and type:
  - class: org.yamcs.ProcessorCreatorService
   args:
     name: "realtime1"
     type: "realtime1"

So it looks much more easy to just create a new instance. The thing with
the new instance is that the data is stored separately so you cannot merge
the two streams in the archive (for example if one stream has some gaps
that can be filled by the other one).

vinnitu, 2019-01-04:
we want to see both graphs simultaneously side by side (or on one graph with different colors)

we tried your configuration with two realtime processors in the same instance and saw lots of messages like  this

```
10:30:00.063 [21] SystemParametersCollector satelite WARNING duplicate value for /yamcs/user/cmdQueue/ops/numCommands
first: name: /yamcs/user/cmdQueue/ops/numCommands genTime: {2019-01-04T08:30:00.063Z} engValue: {0}
 second: name: /yamcs/user/cmdQueue/ops/numCommands genTime: {2019-01-04T08:30:00.063Z} engValue: {0}
10:30:00.063 [21] SystemParametersCollector satelite WARNING duplicate value for /yamcs/user/cmdQueue/ops/numSentCommands
first: name: /yamcs/user/cmdQueue/ops/numSentCommands genTime: {2019-01-04T08:30:00.063Z} engValue: {0}
 second: name: /yamcs/user/cmdQueue/ops/numSentCommands genTime: {2019-01-04T08:30:00.063Z} engValue: {0}
10:30:00.063 [21] SystemParametersCollector satelite WARNING duplicate value for /yamcs/user/cmdQueue/ops/numRejectedCommands
first: name: /yamcs/user/cmdQueue/ops/numRejectedCommands genTime: {2019-01-04T08:30:00.063Z} engValue: {0}
 second: name: /yamcs/user/cmdQueue/ops/numRejectedCommands genTime: {2019-01-04T08:30:00.063Z} engValue: {0}
10:30:00.064 [21] SystemParametersCollector satelite WARNING duplicate value for /yamcs/user/cmdQueue/uhb/state
first: name: /yamcs/user/cmdQueue/uhb/state genTime: {2019-01-04T08:30:00.063Z} engValue: {DISABLED}
 second: name: /yamcs/user/cmdQueue/uhb/state genTime: {2019-01-04T08:30:00.063Z} engValue: {BLOCKED}
```
also we tried to append second datalink to the same instance and saw warnings (in Events section) about broken sequence order (this is obvious)

ideally we would like to be able to choose primary datalink after comparing its quality (less count of warning/errors like above)

xpromache, 2019-01-04:
Ok, I see. I will tryout in the weekend a config with two realtime
processors.

However, that won't solve your problem because you will not be able to see
data from both of them on the same screen. Even worse, now it's difficult
to switch between two different realtime processors due to some missing
functionality in the web displays (it is possible to switch between a
realtime and multiple replay processors).

So I'm thinking now that another solution is to have the xls (MDB) file
loaded twice with different parent subsystem-names. Then all the parameters
will appear duplicated in the MDB, for example:
/DL1/SUB1/PARA1
/DL2/SUB1/PARA1

Is this something you might consider? I can make the mdb config a little
easier (currently you need to make two empty parent sheets with the names
"DL1" and "DL2").

Related to the broken sequence order events: where do they come from? Can
you send a screenshot?
Normally two different data links should not complain about this because
it's very well possible to have a datalink downloading data in parallel
(for example dumping some on-board packets).

vinnitu, 2019-01-04:
[image]

It is works fine if data in links is unique and sequential like this

```
----------------------
dl1  1 2     5 
-----------------------
dl2      3 4   6
-----------------------
```

but in our case data is duplicated in links and can have gaps

```
----------------------
dl1  1 2   4 5 6 
-----------------------
dl2  1 2 3 4   6
-----------------------
```

and even can have offset in time

```
----------------------
dl1  1 2 3 4
-----------------------
dl2    1 2 3 4
-----------------------
```

xpromache, 2019-01-04:
Ok I see but the messages about sequence count discontinuity should be
related to one link only. I.e. the fact that there are two links shouldn't
matter (except that each makes its own message so double the messages).

I think in your case is best to merge somehow the two links eliminating
duplicates.

What kind of data do you receive? Is it like CCSDS packets or something
else?

Do you have some error detection to ensure that bad packets are remove?

vinnitu, 2019-01-05:
Now it is cadu transfer frames with ccsds packets. Here we detect low level errors, reed-solomon correction, etc. Then pass filtered parsed ccsds packets to yamcs. And we have two such handlers (two receivers send data to us through etherner interface udp). But as I mentioned before it is the same data.

xpromache, 2019-01-05:
Ok, so what if we put something in Yamcs to remove duplicates and merge the
two streams into one?

It's not currently part of Yamcs but  I did it in other project and I can
make it part of Yamcs.

nicolae

closed 2019-11-12 by xpromache

vinnitu, 2021-01-26:
Hi! So if I understand this issue solved?

xpromache, 2021-01-26:
Not sure which issue you are referring to, but if I read your very original post I think now you can make two data links and associate them with two different root xtce containers. Those two could be created by loading the same spreadsheet twice under different space systems. So you can have two parameters:
/L1/path/to/parameter1
/L2/path/to/parameter2

vinnitu, 2021-01-26:
Nicolae, let's start again. 

Suppose we have one source of telemetry and many ways to deliver it to endpoint (some radio transmitters and receivers corresponds). In our case we have PC which listen on different udp ports and received same CCSDS frame streams but has gaps(rorate rocket\satelite) and time delays (network)  etc (PC cpu loading i.e.). But finally we want get best of all it. Merge streams in one filling gaps and deduplicating same frames.
(BTW in the nature we have such scenarious here for example https://tools.ietf.org/html/rfc7198 but in our case it is frame count field and algoryth of merging is not clear for me yet)

xpromache, 2021-01-26:
I guess your packets are timestamped by the source and no matter via which
route they come they are the same. Then they will be merged in the archive
and duplicates not stored.

The question is how you want to show them in the realtime display.
Do you expect some paths significantly delayed compared to others? Then
merging them in the realtime display will result in some incoherence. But
if there are no significant delays you can show them as they come.

Btw, we typically extract packets from the frames and store the packets.

vinnitu, 2021-01-26:
We save packets too.
But packet can't be saved if placed in some transfer frames and one of frame was corrupted/missing.
That is why we need merge date from all streams to recover corrupted frames and fill gaps (I can draw example if needed).
And yes after replay raw telemetry data we see some race between streams frame counters.
I think we can't solve this task without some reasonable time frame waiting for frames (no more than 1 second)

vinnitu, 2021-01-26:
one note: it is no needs show telemetry data from each data channel now - only merged\recovered stream

xpromache, 2021-01-27:
Ok, so you want to merge the frames from multiple UDP connections into one
stream. I think currently the only way this can be done is by writing a
custom Data Link (starting maybe from
https://github.com/yamcs/yamcs/blob/master/yamcs-core/src/main/java/org/yamcs/tctm/ccsds/UdpTmFrameLink.java
).

The challenge is indeed if we could do something to Yamcs to allow this
without  a custom data link (or any hand written code).
What kind of frames do you have? TM or AOS?

vinnitu, 2021-01-27:
not TM and not AOS. It is custom based on AOS. No matter what kind of frame. In frame must be counter. Let's consider AOS.

xpromache, 2021-01-27:
Currently the only way is to make your own data link to do the frame
buffering and sorting.

I think some StreamSQL with windows could do this kind of thing but we need
some extra development in Yamcs for that. See for example:
https://docs.sqlstream.com/sql-reference-guide/select-statement/window-clause/

Yamcs supports stream windows but not order by.
In addition, one would need a generic mechanism to put the frames received
via UDP to streams.

Something like this:

UDP receiver \
UDP receiver   --- stream[window ordering] -> frame processing
UDP receiver /

Maybe it will be possible in the future, for the moment the only solution I
see is you creating your own data link.

nicolae

vinnitu, 2021-03-12:
it is has name Best Source Selector