yamcs/yamcs #305: wrong type of start, stop

vinnitu, 2019-01-09:
in doc we have https://www.yamcs.org/docs/http/Get_Tag/

[image]

but really 

[image]

must be int64

[image]

xpromache, 2019-01-09:
Looks to be originating from limitations in Javascript which uses doubles
for all the numbers:
https://github.com/protocolbuffers/protobuf/issues/2679
https://stackoverflow.com/questions/9643626/does-javascript-support-64-bit-integers

In this particular case we should export the timestamps in UTC strings
because those numbers are difficult to deal with from outside Yamcs.

xpromache, 2019-01-09:
Somewhere in protobuf 3  a Timestamp type has been introduced (and is backward compatible with protobuf 2 which we are using):
https://github.com/protocolbuffers/protobuf/blob/master/src/google/protobuf/timestamp.proto

The protobuf to json conversion routine will convert this automatically to a string whereas the binary representation will do the normal protobuf conversion of the message. 

Therefore it looks like a good idea to start using this Timestamp type instead of manually converting the timestamp or sending two fields (one integer and one string) like we do now for example in parameter value class:
https://www.yamcs.org/docs/http/Get_Parameter_Value/

closed 2019-11-12 by xpromache