yamcs/yamcs #306: how to use unix time epoch

vinnitu, 2019-01-10:
I see in ParameterTypeProcessor.java 

[image]

What I need to configure to use UNIX epoch time?

xpromache, 2019-01-11:
sorry, where do you want to use the unix time epoch?

vinnitu, 2019-01-11:
inside ccsds packet in secondary header

Vitaliy136, 2019-01-23:
Hello! 
I am trying do the same.
I put unix time stamp in coarse-time in secondary header space packet.
I modified xls config like this
[image]
[image]
Is it correct?
But seems calibration does not work because in yamcs-web i see shifted time (2029)

xpromache, 2019-01-23:
Can you tell me where you see the shift (a screenshot would be appreciated)?

Do you have something in Calibration->calib2 column for the time_subsec?  (I cannot see it in the screenshot you provided, it kind of looks empty). That should contain the scaling factor (from whatever resolution to seconds).

Vitaliy136, 2019-01-23:
I see time shift in yamcs-web for parameters

[image]

Scaling factor does not affect the coarse-time to make an offset for 10 years (as I understand it).
I generate time for packets in this way
dt = new Date()
coarse_time = floor(dt / 1000);
fine_time = floor(((dt % 1000) * 256 / 1000));

xpromache, 2019-01-24:
The timestamp you see there is not the value of the parameter (computed according to the spreadsheet definition) but the timestamp associated. Currently the timestamp associated is coming from the so called PacketPreprocessor which if you didn't change our simulator configuration is the IssPacketPreprocessor which interprets always the timestamp as offset from GPS Epoch.

This timestamp is derived when the packet is injected into Yamcs, before the XTCE processing (i.e. the spreadsheet) which happens afterwards. This is a current limitation of Yamcs but it's difficult to overcome because we need that timestamp do archive the packet even without processing it.

Now, I can put an option on that IssPacketPreprocessor to interpret time differently but my question to you is: why do you use the IssPacketPreprocessor? Is there some need to be compatible with something? Where are your packets coming from actually?

xpromache, 2019-01-24:
> I do not configure PacketPreprocessor at all
>
> services:
>   - class: org.yamcs.tctm.DataLinkInitialiser
>   - class: org.yamcs.archive.XtceTmRecorder
>   - class: org.yamcs.archive.ParameterRecorder
>   - class: org.yamcs.archive.AlarmRecorder
>   - class: org.yamcs.archive.EventRecorder
>   - class: org.yamcs.archive.ReplayServer
>   - class: org.yamcs.archive.IndexServer
>   - class: org.yamcs.parameter.SystemParametersCollector
>     args:
>       provideJvmVariables: true
>       provideFsVariables: true
>   - class: org.yamcs.ProcessorCreatorService
>     args:.
>       name: "realtime"
>       type: "realtime".
>   - class: org.yamcs.archive.CommandHistoryRecorder
>   - class: org.yamcs.parameterarchive.ParameterArchive
>     args:
>       realtimeFiller:.
>         enabled: false
>       backFiller:
>         enabled: true
>         warmupTime: 60
>
>
> dataLinks:
>   - name: tm_realtime
>     class: org.yamcs.tctm.UdpTmDataLink
>     args:
>       stream: tm_realtime
>       host: localhost
>       port: 10015
>
>
> mdb:
>   # Configuration of the active loaders
>   # Valid loaders are: sheet, xtce or fully qualified name of the class
>   - type: "sheet"
>     spec: "mdb/simulator-ccsds.xls"
>     subLoaders:
>       - type: "sheet"
>         spec: "mdb/landing.xls"
>
> #Configuration for streams created at server startup
> streamConfig:
>   tm:
>     - name: "tm_realtime"
>   event: ["events_realtime"]
>   param: ["pp_realtime", "sys_param"]
>   alarm: ["alarms_realtime"]
>
>
> What I need to set up ?

If you don't configure it, the default one IssPacketPreprocessor
<https://www.yamcs.org/yamcs/javadoc/index.html?org/yamcs/tctm/IssPacketPreprocessor.html>
is
used (this from the time when Yamcs was only used for ISS):
https://www.yamcs.org/docs/server/UDP_TM_Data_Link/

But can you tell me where your packets are coming from?

As I said, I propose one of the two:
1. add an option for the IssPacketPreprocessor
<https://www.yamcs.org/yamcs/javadoc/index.html?org/yamcs/tctm/IssPacketPreprocessor.html>
 to allow to use Unix time.
2. make another packet pre-processor that is very simple and has more
options (if we don't need to follow any CCSDS or other standards).

I prefer 2. because in ISS packets the time is supposed to be GPS so if we
have an option to change that time into UNIX we kind of violate the packet
definition.

nicolae

Vitaliy136, 2019-01-24:
I created simple packets generator in node js for tests. Thank you

xpromache, 2019-01-24:
ok, then please wait until Monday, I will commit a
GenericPacketPreprocessor that can work with Unix time.

I propose this structure:
packet size (16 bits) - the size of the packet data to follow in big endian
packet data

I will have an option to read a 4 bytes sequence number and a 8 byte
timestamp from a user defined offset.
The 8 bytes timestamp is millisec from 1970 (like java
System.currentTimeMillisec()).

Is that ok with you?
Do you need a checksum (normally not required if you only have ground
communications)?

nicolae

xpromache, 2019-01-28:
Please have  a look here:
https://github.com/yamcs/yamcs/blob/master/yamcs-simulation/etc/yamcs.simulator.yaml
and search for tm2_realtime.

It should do what you wanted.

vinnitu, 2019-01-29:
Ideally I see some GenericCcsdsPacketPreprocessor which has optional secondary header, optional checksum, optional type-id, optional packet-id, time stamp (one of described in ccsds recommendation) setting in instance yaml config file and setting epoch in xls. So if I need only 2 type space packets - I don't use type-id, packet-id at all (but apid from primary header as switch), etc.

vinnitu, 2019-01-29:
Or such situation when packet doesn't have time stamp at all - can yamcs set system timestamp when receive data?

xpromache, 2019-01-29:
Yes, that should be possible, I will put some more options on the
GenericPacketPreprocessor next weekend.
I think we don't need to care about the CCSDS header too much as long as
you can specify where to extract the necessary fields (sequence count,
timestamp) from.

closed 2019-11-12 by xpromache