yamcs/yamcs #309: unsubscribe bug

vinnitu, 2019-01-24:
Hello!

0. F12 -> Network -> WS
1. Select Displays
2. Create display
3. Next
4. Save
5. +Add (Search any realtime parameter)
6. Add
7. Save
8. see in WS tab - running data
9. Layouts
10. see in WS tab - still running data!!!

vinnitu, 2019-01-24:
even more - if you go back and select in Displays .par file again - you will see copy of subscribed data!!

fqqb, 2019-01-28:
You are seeing the same values multiple times in the front-end itself? Or only on the websocket connection?

You are right that on the websocket connection it may lead to double data. This is something that should be cleaned up. The historic reason is that yamcs-web works with one connection and before we unsubscribe a parameter, we should be sure that it is not used by any other component. The 'simple' solution we had for this was to always add to the subscription and never delete. If you refresh the browser, or navigate to the homepage, you get a fresh connection anyway.

Only recently we added functionality to manage distinct subscriptions identifiers over the same websocket connection, so we could improve this behaviour now.

vinnitu, 2019-01-28:
I see it only in websocket tab

closed 2019-02-26 by vinnitu