yamcs/yamcs #312: lightblue

vinnitu, 2019-01-28:
Hello, what does lightblue mean?

[image]

xpromache, 2019-01-28:
The values shown there are (average, min, max) for each equally sized
interval where the length of the interval is determined by the total time
of the graph divided by the interval count.
This is the call that is being used:
https://www.yamcs.org/docs/http/Get_Parameter_Samples/

The average is shown in dark blue whereas the min/max are shown as a
vertical line in light blue.

vinnitu, 2019-01-28:
As I understand it is {errorBars: true} option in dygraph http://dygraphs.com/options.html#Error%20Bars

fqqb, 2019-01-28:
it uses dygraphs customBars using (min, avg, max) for each sample interval.

closed 2019-11-12 by xpromache