yamcs/yamcs #320: Layouts list

dvissert, 2019-02-20:
Hello! Recent updates made it impossible to get available layouts list. Steps to reproduce:
1. Create some display (parameter table for /YSS/SIMULATOR/BatteryVoltage1 in my case)
2. Create new layout with name 42, add display and save changes
3. Go to /monitor/layouts?instance=simulator with `You did not yet save any layouts.` message
4. Direct link to /monitor/layouts/42?instance=simulator still shows saved layout

fqqb, 2019-02-20:
I did a change such that displays (and layouts) are by default stored and saved to global storage (as opposed to instance storage). This way they can be used in any instance.

But it looks like I made a mistake on the page for an individual layout because it still looks for a layout in the instance storage.

I'll make a fix shortly, but meanwhile you can probably use this workaround: in `etc/yamcs.yaml` modify the HttpServer configuration to specify that the `displayScope` is INSTANCE. (the default is now GLOBAL):

```yaml
services:
  - class: org.yamcs.web.HttpServer
    args:
      webRoot: lib/yamcs-web
      port: 8090
      webConfig:
        displayScope: INSTANCE
```

This should revert you to the old behaviour, and allow you to list your layouts.

dvissert, 2019-02-20:
Switching `displayScope` worked, thanks!

fqqb, 2019-02-21:
Fix merged to master. This new configuration option is now also documented:
https://www.yamcs.org/docs/server/HTTP_Server/

closed 2019-02-21 by fqqb