yamcs/yamcs #322: Parameter_Updates

vinnitu, 2019-02-21:
I see it dev console command to subscribe on Altitude

[1,1,4,{"parameter":"subscribe","data":{"id":[{"name":"/YSS/simulator/Altitude"}],"abortOnInvalid":false,"sendFromCache":true,"subscriptionId":-1,"updateOnExpiration":true}}]

but [here](https://www.yamcs.org/docs/http/Parameter_Updates/) I don't see description of this case, I mean id in data

So what is the properly way to subscribe?

fqqb, 2019-02-21:
it should be `id`, i've adjusted the page. thanks

closed 2019-02-21 by fqqb

vinnitu, 2019-02-21:
ooook...

but, how to properly make unsubscription?

[image]

as you can see
```
[1,3,3,{"et":"STRING","msg":"Cannot find field: id in message yamcs.protobuf.NamedObjectList"}]
```