yamcs/yamcs #323: how to properly make unsubscription

vinnitu, 2019-02-22:
ooook...

but, ?

[image]

as you can see
```
[1,3,3,{"et":"STRING","msg":"Cannot find field: id in message yamcs.protobuf.NamedObjectList"}]
```

_Originally posted by @vinnitu in https://github.com/yamcs/yamcs/issues/322#issuecomment-466096413_

fqqb, 2019-02-25:
that's a bug indeed. I'll work on a fix.
Both subscribe and unsubscribe used to use a field `list`, which was later renamed to `id` (apparently not successfully).

Possibly you can work around it by changing "id" into "list" (on unsubscribe only), but i didn't try.

xpromache, 2019-02-26:
Fixed

closed 2019-02-26 by xpromache