yamcs/yamcs #337: twosComplement for float

vinnitu, 2019-05-22:
Hello!

How can we describe in config data float(32) type which received on space-packet generator side from ABCD into CDAB form?

xpromache, 2019-05-22:
Not possible but you can make a custom decoder for it. It only works in
Java, if you are not familiar with java development you can try to follow
this (I haven't tried it myself):

1. Create a class in a directory inutil (this is the package name you can
change it as you want):
inutil/MyFloatDecoder.java :

package inutil;

import org.yamcs.algorithms.AlgorithmExecutionContext;
import org.yamcs.parameter.Value;
import org.yamcs.utils.BitBuffer;
import org.yamcs.utils.ValueUtility;
import org.yamcs.xtce.Algorithm;
import org.yamcs.xtce.DataEncoding;
import org.yamcs.xtceproc.AbstractDataDecoder;

public class MyFloatDecoder extends AbstractDataDecoder {
    public MyFloatDecoder(Algorithm a, AlgorithmExecutionContext c) {
    }

    @Override
    public Value extractRaw(DataEncoding de, BitBuffer buf) {
        int x = (int) buf.getBits(32);
        int y = (x<<16)|(x>>>16); //check that this is what you want!!!
        float f = Float.intBitsToFloat(y);
        return ValueUtility.getFloatValue(f);
    }
}

2. compile
javac -cp "<YAMCS_LIB_DIRECTORY>/*" inutil/MyFloatDecoder.java
Where YAMCS_LIB_DIRECTORY is the lib directory of yamcs containing the
yamcs jars are (e.g. /opt/yamcs/lib).

3. create a jar out of your compiled class
jar cvf MyFloatDecoder.jar inutil/MyFloatDecoder.class
and move it to the <YAMCS_LIB_DIRECTORY>

4. edit the spreadsheet and create a new data type (in the DataType sheet)
MyFloatEncoding float float custom(algo_my_float_encoding)

add also a line in the Algorithm sheet
algo_my_float_encoding Java inutil.MyFloatDecoder

and finally define your parameters in the ParameterSheet to have the type
MyFloatEncoding.

(you can look for an example in the refmdb.xls)

5. restart Yamcs

6. write back if it is not working (or even if it is)

7. Have fun!
:)

nicolae

PS: I'll answer later to the issue you posted yesterday.

xpromache, 2019-05-22:
Not possible but you can make a custom decoder for it. It only works in
Java, if you are not familiar with java development you can try to follow
this (I haven't tried it myself):

1. Create a class in a directory inutil (this is the package name you can
change it as you want):
inutil/MyFloatDecoder.java :

```java

package inutil;

import org.yamcs.algorithms.AlgorithmExecutionContext;
import org.yamcs.parameter.Value;
import org.yamcs.utils.BitBuffer;
import org.yamcs.utils.ValueUtility;
import org.yamcs.xtce.Algorithm;
import org.yamcs.xtce.DataEncoding;
import org.yamcs.xtceproc.AbstractDataDecoder;

public class MyFloatDecoder extends AbstractDataDecoder {
    public MyFloatDecoder(Algorithm a, AlgorithmExecutionContext c) {
    }

    @override
    public Value extractRaw(DataEncoding de, BitBuffer buf) {
        int x = (int) buf.getBits(32);
        int y = (x<<16)|(x>>>16); //check that this is what you want!!!
        float f = Float.intBitsToFloat(y);
        return ValueUtility.getFloatValue(f);
    }
}
```

2. compile
javac -cp "<YAMCS_LIB_DIRECTORY>/*" inutil/MyFloatDecoder.java
Where YAMCS_LIB_DIRECTORY is the lib directory of yamcs containing the
yamcs jars are (e.g. /opt/yamcs/lib).

3. create a jar out of your compiled class
jar cvf MyFloatDecoder.jar inutil/MyFloatDecoder.class
and move it to the <YAMCS_LIB_DIRECTORY>

4. edit the spreadsheet and create a new data type (in the DataType sheet)
MyFloatEncoding float float custom(algo_my_float_encoding)

add also a line in the Algorithm sheet
algo_my_float_encoding Java inutil.MyFloatDecoder

and finally define your parameters in the ParameterSheet to have the type
MyFloatEncoding.

(you can look for an example in the refmdb.xls)

5. restart Yamcs

6. write back if it is not working (or even if it is)

7. Have fun!
:)

nicolae

PS: I'll answer later to the issue you posted yesterday.

closed 2019-11-12 by xpromache