yamcs/yamcs #358: Link thread dies in XtceTmExtractor if exception has null message

merose, 2019-10-25:
In XtceTmExtractor.processPacket(), if an exception is thrown with a null message, then pdata.eventProducer.sendWarning() is called with a null pointer, causing the thread to die.
Better to use Exception.toString() instead of Exception.getMessage() (which can return null).

xpromache, 2019-10-26:
Thanks for the report and the fix. What is causing the NPE in the first place?
Is it a bad XTCE file? I think we should also fix that.

merose, 2019-10-26:
I'll have to try to reproduce it, but I think it was a NullPointerException. It was a RuntimeException of some sort that did not have a message, and it was caused by a malformed packet. (There are quite a few malformed packets in the LADEE data we're using, not sure why. That is, the packet did not match the XTCE definition.) The XTCE seems fine.

The symptom I saw previously was that the TM data link would die with no message in the log. I changed the code to print a stack trace and that's when I saw that the exception had no message. (There are many places in the code where Exception.getMessage() is used. Probably those should either be 1) changed to Exception.toString() (you get the class name prepended to the message, but that doesn't clutter the log much), or 2) checked to make sure that the code using the value of Exception.getMessage() is safe if the result is null.)

commit 7ae0de9c (2021-04-21): Eliminate usages of some of Exception.getMessage()

Should fix #358
closed 2021-04-21 by xpromache