yamcs/yamcs #360: Cannot change command queue state in web UI (4.10.3)

merose, 2019-10-25:
Affects release 4.10.3

To reproduce:

1. In the YAMCS web GUI, go to Commanding > Command Queues.
2. Press the 3-dot icon at the right of the "Default" queue and select Change mode > Accept all.

Expected result: Mode of default queue changes to "Accept all".

Actual result: Mode of default queue is unchanged, and there is an error message on the console:

    <6>Method POST not allowed for URI: '/api/processors/ladee/realtime/cqueues/default'
    <4>497366b3 POST /api/processors/ladee/realtime/cqueues/default 400

According to the web services API, a request method of PATCH should be used to change the queue state instead of POST. (See https://www.yamcs.org/docs/http/Edit_Command_Queue/)

fqqb, 2019-10-26:
Thanks for the report,

This issue has been fixed on master a few days ago, see: 

https://github.com/yamcs/yamcs/blob/0b7202b3657796c259ef98c4476c23c7b041ea04/yamcs-web/packages/client/src/InstanceClient.ts#L244

It will be in the next release, along with some other updates to that view (UI visibility on queue order, user/group filters, and command significance filters). I'm finishing another pr for that.

If you want this particular issue fixed earlier you can try to manually rebuild the web sources in yamcs-web from master.

merose, 2019-11-07:
I've verified that this is fixed in 4.10.4. However, see also #365.

closed 2019-11-07 by merose