yamcs/yamcs #361: Should support <xi:include> in XTCE configuration (4.10.3)

merose, 2019-10-27:
It would be really nice to support <xi:include> in XTCE configurations, in order to make composition of multiple <SpaceSystem> definitions possible, as it is for Excel-based configurations.

Unfortunately, the StAX parse in the JDK does not support <xi:include>, so it would have to be implemented in a filtering XMLEventReader implementation.

xpromache, 2019-10-27:
Thanks for the patch.

Do you know it's possible in the mdb definition to add multiple XML files, using subLoaders like done here:
https://github.com/yamcs/yamcs/blob/master/simulation/src/main/yamcs/etc/yamcs.simulator.yaml

You can also combine XML with xls (or with other loader).
It's also possible to cross reference from one to another, and the resolving will be done after all are loaded.

merose, 2019-10-27:
Yes, I saw that, too. There are advantages to both methods. Advantages of Xinclude are 1) if you already know it you don't need to read the manual, and 2) it works at the lexical level rather than at the SpaceSystem level. That is, you can split the SpaceSystem up arbitrarily, putting commands in a separate XML file, say. But the "subLoader" system allows you to compose arbitrarily, which will be an advantage at times.

AdamElliott-AB, 2020-03-26:
I would like to see this feature as well