yamcs/yamcs #365: Cannot change state of first command queue out of "HOLD"

merose, 2019-11-07:
Environment: YAMCS 4.10.4, Mac OS X 10.14.6, Safari 13.0.2

It appears that the flashing "HOLD" indication on queues in the HOLD state makes it difficult or impossible to change the state of the first command queue in the list. To reproduce:

1. Configure at least two queues, all enabled.
2. Start YAMCS.
3. Go to the Commanding > Command Queues screen.
4. Click on the three dots to the right of the first queue, and select Change action > Hold.

Expected and actual result: Queue changes state to the flashing "HOLD" indicator.

5. Click on the three dots to the right of the first queue.

Expected result: The "Change action" popup appears and waits for you to select a new state.

Actual result: the "Change action" popup appears, but then disappears. If you're very fast, you may be able to change the queue state, but I was never successful.

Note: I have reproduced this in Chrome. It does not affect the popup functionality for the other command queues, only the first.

Conjecture: The flashing "HOLD" indication is changing the focus, causing the popup to disappear. Unclear why it affects only the first queue, however.

fqqb, 2019-11-07:
The blink effect is only a CSS animation though.
I can only try later, but I suspect that maybe the queue where you are observing this, is getting updated for some other reason (for example because commands are going through it).

A bug nonetheless, i'll try to reproduce and fix soon.

merose, 2019-11-07:
No, this was when no commands were being sent. But you're right, it's unrelated to the HOLD state. I'll update the description. It only affects the first queue in the list.

fqqb, 2019-11-08:
Can it be that you are using the property "stateExpirationTimeS" on that queue's configuration? I was able to reproduce your description, but only when using this property.

When changing a queue's action, the use of this property causes 1 hz updates on the live data connection (with progress indication). These were causing the row for that queue to be re-rendered (thereby closing the popup).

This behaviour is fixed, as part of pr #366

merose, 2019-11-08:
Yes, I had copied this from etc/command-queue.yaml.sample, but edited it to change to use "minLevel" to avoid the new warning, and moved it to be the first queue.

    ops-critic:
      state: enabled
      minLevel: watch
      stateExpirationTimeS: 300

I don't need an expiration time, so I'll remove it, for now. Thanks for the fix!

closed 2019-11-12 by xpromache