yamcs/yamcs #368: yamcs server fails to start after install on Ubuntu 16

trgarvey, 2019-11-12:
After installing on Ubuntu 16, I get the following error when trying to start yamcs server using "service yamcs-server start":

Nov 12 09:30:14 GDS1Ubuntu su[16043]: Successful su for yamcs by root
Nov 12 09:30:14 GDS1Ubuntu su[16043]: + ??? root:yamcs
Nov 12 09:30:14 GDS1Ubuntu su[16043]: pam_unix(su:session): session opened for user yamcs by (uid=0)
Nov 12 09:30:15 GDS1Ubuntu yamcs-server[16035]: Starting Yamcs...   Yamcs failed to start. See log file for details.
Nov 12 09:30:15 GDS1Ubuntu systemd[1]: yamcs-server.service: Control process exited, code=exited status=1
Nov 12 09:30:15 GDS1Ubuntu systemd[1]: Failed to start LSB: Yamcs Mission Control.
Nov 12 09:30:15 GDS1Ubuntu systemd[1]: yamcs-server.service: Unit entered failed state.
Nov 12 09:30:15 GDS1Ubuntu systemd[1]: yamcs-server.service: Failed with result 'exit-code'.

xpromache, 2019-11-12:
Did you check in /opt/yamcs/log/yamcs-server.out for any error?

Did you install also the yamcs-simulation package? That includes some config files without which the server will not start.

trgarvey, 2019-11-12:
I did not install the YAMCS simulation package.  This is probably the issue.
According to the logflie it did not find the yamcs.yaml config file.  What is odd is that the entire /opt/yamcs/etc directory is empty.

I installed by converting the .rpm package to a deb package using Alien since I am on Ubuntu.  It did seem to install successfully but it is strange that the /opt/yamcs/etc directory has no files in it.

xpromache, 2019-11-12:
The original rpm should have some yaml.sample files in the etc directory.
The yamcs-simulation contains files that allow to start Yamcs in a
simulator configuration. I think it's easier to start like that to see what
Yamcs is doing.

trgarvey, 2019-11-12:
Successful start after installing the simulation!  However, the web service does not appear to be running on port 8090 (unless the simulation configuration does not start the web service).

xpromache, 2019-11-12:
Indeed, there seems to be some issue with the simulation config.
As a workaround, please comment out in yamcs.yaml the yamcs-web section
(with all underneath) and restart the yamcs-server.
We will figure out tomorrow what the problem is.

trgarvey, 2019-11-12:
I launched yamcs-studio and there does seem to be both events and telemetry sources from the simulator available there.  So perhaps it is just the web interface config...

xpromache, 2019-11-12:
As I said there is a problem with the web configuration so just comment out
that section and the web interface should work in the default
configuration.

fqqb, 2019-11-13:
It should suffice to comment out only the `staticRoot` property in `/opt/yamcs/etc/yamcs.yaml`. We use this property during development. For RPM installations it should not be used.

A fix for this is included in pr #369 - for the next release.

Thanks for reporting this.

closed 2020-05-07 by xpromache