yamcs/yamcs #370: CFDP web interface does not indicate file chosen for uplink

trgarvey, 2019-11-19:
In the server web interface, when a file is chosen for uplink from a bucket, the "Start Transfer" button will activate when a file is chosen, but there is no confirmation which file has been chosen.  The filename is not highlighted once it is clicked, and there is no string showing the file chosen.  Thus it is easy to click on the wrong file and have no indication what is uplinked.

fqqb, 2019-11-20:
That can be done yes, it'd be useful also as a way to show the full path of the selected file.
I'll look at it when i can.

btw, i can't reproduce this highlight issue on my copy of Firefox.

trgarvey, 2019-11-20:
It may only be on Ubuntu Linux with Firefox, or even just the computer I am using.  It affects any text that is "highlighted" in grey within the YAMCS Web Interface.  It may be specific to my situation.  Using a different browser seems to be a good solution.  Thanks for taking the suggestion of a string indicator for the file selection though.

xpromache, 2021-03-29:
I think this has been fixed with the latest rework of the CFDP.

closed 2021-03-29 by xpromache