yamcs/yamcs #373: Some XTCE aggregate parameters not returned by /api/mdb/<instance>/parameters

merose, 2020-01-23:
(This is somewhat vague because I have not yet reduced this to a small XTCE input file. When I have time I'll add more details.)

I would expect that the HTTP API /api/mdb/<instance>/parameters would return all parameters. However, I have some parameters that use nested aggregate types. These appear not to be returned by that API call. However, when using /api/mdb/<instance>/parameters/<name>, I can get the information on those parameters. Further, they appear in the YAMCS UI in the MDB > Parameters tab.

merose, 2020-01-23:
I'm going to close this. It may be that adding "details=yes" solves the problem. (BTW, should document that in the on-line manual. I had to read the source to find that.)

closed 2020-01-23 by merose

fqqb, 2020-01-24:
This API call should indeed return all parameters (details or no-details).

However for aggregate (and array) parameters, only the top level definition is really considered to be a parameter (in the XTCE sense). The member structure is part of the type definition of that parameter.

The call defaults to no-details, because it's only supposed to give a quick listing of existing parameters (the difference in transfer size can be significant depending on your MDB).

I've updated the online docs with this "details" option. Note that the latest API documentation is currently best viewed here: https://github.com/yamcs/yamcs/tree/master/docs/http-api. The API docs on yamcs.org are getting a bit outdated as we've been preparing a new website in parallel (about a month away).