yamcs/yamcs #377: Enabling the realtime parameter archive filler causes Yamcs to ignore SIGINT/SIGHUP

merose, 2020-02-13:
None of the example Yaml configuration files enable the realtime parameter archive filler. Is there a reason _not_ to use the realtime filler?

Also, if the realtime filler is enabled, the Yamcs server no longer responds to SIGINT (=Ctrl-C) nor SIGHUP. Instead, I've had to use SIGKILL, which is undesirable, of course.

fqqb, 2020-02-27:
Undesirable indeed. It would have stopped after the flushInterval (default 300 seconds).
Issue is fixed on master, next release should be tomorrow.

(the bug was introduced when we started doing graceful shutdowns on services)

fqqb, 2020-02-27:
Reopening for the first part of the question (leaving that to @xpromache)

xpromache, 2021-03-29:
Finally spent some time to fix the realtime filler. Now it is able to also create segments of max size like the back filler. In addition the compression and writing to archive is done in background threads so it should be reasonable performant on a multicore machine (meaning that it will not block the realtime processing).
It also works as a parameter cache.

If all you get is realtime data, there is no reason anymore to use the backfiller on a regular schedule.
(however the backfiller can still be invoked manually if some old data is injected in the archive, or if a reprocessing is needed for example due to a change in MDB)

The quickstart example will be changed to use the realtime filler.

closed 2021-03-29 by xpromache

merose, 2021-04-05:
Under what conditions is "all you get is realtime data" true? Is this true if all data comes in through data links? (No matter the packet generation time.) Or if packets are received older than a threshold time not processed by the realtime filler? (For example, there's a problem with an SLE link causing you to lose some frames/packets during a time period, and you fill in the data using the offline-complete service a day later.)

The reason I ask is that we're seeing that the backfiller re-runs algorithms, while the realtime filler appears not to. We want to make sure that algorithm results make it into the archive. We'd prefer not to run the backfiller if the realtime filler will be sufficient, but we want to understand the difference. Thanks!

xpromache, 2021-04-05:
The realtime filler writes all the data that passes through the realtime
processor. In practice that's the data you see on your screens during
realtime operations. Generally you don't want to see old data on the screen
and in any case not mixed with new data because that would be really
confusing.

So when you get data recorded at a ground station or recorded on-board, you
insert that into the stream (raw) archive without passing it through a
processor (see the tm_dump tm streams on the simulator example).
Then to process the data for the parameter archive you use the backfiller -
that starts a replay processor which converts raw data (packets) to
parameters for the Parameter Archive.

I tried to capture the process in this diagram:
https://docs.yamcs.org/yamcs-server-manual/general/architecture/

The algorithms are part of the processor be it realtime or the replay one
launched by the back filler(in fact the processors made by the backfiller
are with the type "ParameterArchive" in processor.yaml - so you can
configure there what kind of services they have (e.g. with algorithms or
without)).

And one more note to make it even more confusing: the stream archive can
also store parameters (everything that goes in the "pp" table) - those are
not really processed when being moved to the parameter archive (although
raw to engineering value can still be done I think).
These parameters are data typically received in realtime that are not
packets - for example the system parameters (everything under /yamcs
namespace) or  data produced by external test equipment, or what we call
local parameters set via API.

So I think in the usecase of monitoring a flying spacecraft/instrument, you
almost always need a backfiller.But it can be only manually invoked after
injecting some past data or can automatically trigger with the
monitorStreams option. BTW I spent also some time to documented all the
options:
https://docs.yamcs.org/yamcs-server-manual/services/instance/parameter-archive-service/

The "all you get is realtime data" is mostly during development and tests
when you are connected directly to the target device and there is no
intermediate storage anywhere (any data that is lost by Yamcs due to
disconnection or other reasons, is lost forever).

merose, 2021-04-06:
Thanks. Let me try to paraphrase your comments to see if I understand.

* All data from a data link connected to the `tm_realtime` stream will be processed by the realtime processor, which means that it will be archived by the realtime filler. (This actually describes our mission, because our spacecraft does not store data when it cannot communicate with the ground.) In this situation, the backfiller can be disabled. The backfiller might be run on demand if the calibration changes, for example, but otherwise is not used.
* If data is stored on the spacecraft and downlinked in a large dump, it is convenient to configure a data link to receive that data which uses the `tm_dump` stream instead. This will not be processed by the realtime processor. Instead, the parameters must be archived by the backfiller. The backfiller can also be set to _disabled_ and run on demand in this case after data is dumped.
* A third case might be the situation where most all data comes in through realtime links, but occasional gaps in telemetry might be filled in afterward, such as from an _offline complete_ SLE service. In this case the data link should feed the `tm_dump` stream, and the backfiller should be run manually afterward over the time range in order to fill in the parameter archive from that new data.

xpromache, 2021-04-07:
Yes, I think your understanding is correct. Yamcs has its roots in ISS
operations and the third case is the modus operandi there - you get good
coverage via the TDRS relay satellites but the coverage is not continuous;
so ISS has TM recorders at station level and each AOS, you get two data
streams  - a realtime stream and a dump of what has been recorded during
the previous LOS (or even LOSes hours ago if the link was busy and they
couldn't do the dump). Most payload/instruments are not aware of the
AOS/LOS they keep producing data continuously.

But the AOS/LOS transition is never a clear cut so they always start the
recorders before the LOS and stop them after the AOS which means that the
next dump will overlap a bit with the data received via realtime. And to
further complicate the issue you get S-band and KU-band data each with
different AOS/LOS periods.

Since the AOS/LOS transition goes on every 10 to 30 minutes, we do not
operate the realtime filler, we do everything with the backfiller. The
realtime filler would produce an archive full of gaps which would be
overwritten anyway by the backfiller.

One important thing to realize is that the way the parameter archive is
implemented, you cannot fill one single data point (corresponding to one
timestamp). We split the archive based on intervals of approximately 139
minutes (2^23 milliseconds exactly) and we need to rebuild the whole
interval if something changes inside. We do that in order to be able to
compress data in one interval  - for example if a parameter doesn't change
its value at all in those 139 minutes, it is compressed to almost nothing.

Many times people have algorithms that depend on parameters other than the
ones from a current packet. So in order to be able to build a proper state
you need to replay data for some time. Just converting one single packet to
parameters does not make sense. That is why the "limitation" with the
parameter archive intervals has been considered acceptable - even if the
data was stored differently you would still need to process a whole stream
of packets, you cannot just convert one packet to parameters in isolation.

commit d0046410 (2021-03-29): fixed the realtime parameter archive filler

Fixes #377
commit 70232280 (2021-03-29): fixed the realtime parameter archive filler

Fixes #377