yamcs/yamcs #383: Blank documentation pages

AdamElliott-AB, 2020-03-04:
Some documentation pages are now empty.
https://yamcs.org/docs/yamcs-server-manual/mdb/command-definitions/
https://yamcs.org/docs/yamcs-server-manual/mdb/parameter-definitions/

fqqb, 2020-03-04:
thanks for the report, they are placeholders introduced after a recent documentation restructuring.
I'll prepare new content in the coming days.

AdamElliott-AB, 2020-03-04:
Thank you for the quick response. I thought I saw content there before, but perhaps I was mistaken.

closed 2020-03-04 by AdamElliott-AB

fqqb, 2020-03-04:
It could be you're referring to this content: https://yamcs.org/docs/yamcs-server-manual/mdb/loaders/sheet/

This details how to encode the mission database using a spreadsheet. It's the best documented option, because we've had this around for a long time.

But I think we should have some general MDB documentation - unattached to any of the input formats.