yamcs/yamcs #390: How to provide runtime configuration to commanding?

taguan, 2020-04-01:
Hi,

I don't find a way to provide some kind of runtime configuration (directives) when commanding. My goal is to specify a virtual channel ID and a mapId when sending a command. This information needs to be taken into account by my data link when sending the command.

If I specify this as a command argument in the MDB, this is going to be encoded in the packet, and that's not my goal.
If I use static configuration, I can only specify that some commands are associated to a given default vcId or mapId, but it doesn't give me a solution to change that value at runtime when sending a command.

Do you have an approach to suggest?

xpromache, 2020-04-02:
Hello,

We have added recently the possibility to specify some additional command options when sending a command but the functionality is not yet propagated to all clients especially since we are hesitating on the proper way to do it.

See here:
https://github.com/yamcs/yamcs/blob/master/yamcs-api/src/main/proto/yamcs/protobuf/commanding/commanding.proto

the message CommandOptions that can be attached when sending a command. The "attributes" part of that message is supposed to be used for the kind of thing you want.
See here how an similar attribute "cop1Bypass" can be used in a data link:
https://github.com/yamcs/yamcs/blob/30f891eac819e9d1595c939b4477c86d8e09f7dc/yamcs-core/src/main/java/org/yamcs/tctm/ccsds/Cop1TcPacketHandler.java

Currently the python client supports setting the Verification part and no client supports the attributes part (of course you can always send it as part of the REST request (using e.g. curl)). 

It should be added in the yamcs-web, Yamcs Studio and python client in the coming weeks.
One problem we have to figure out is how to tell to a client which extra attributes can be set since this depends on the data link configuration.

Currently we do not have a simple solution for that, we are still thinking what the best option is; as you say it could be part of the MDB but we didn't find a XTCE way to do that, so maybe some sort of extension.However it does not sound right that the MDB (which is a system description) will change depending on the link (which says how to transport data to the system) configuration.

taguan, 2020-04-02:
Hi,

Thanks for your answer, you have perfectly understood the issue. The proposed solution is the one I am expecting.

I missed the fact that PreparedCommand defines "attributes" that are not the same as those of the MetaCommand. That's what I need.
The use of the REST API is OK for me, even if would indeed be nice to have support in the various clients.

For the web client, perhaps that it could simply be a configured "string" list giving the name of the extra attributes? I don't think there is a need for something more sophisticated than that.

As a workaround, I encoded my transport protocol in the MDB, but I don't like that, for the same reason as the one you mention.

By the way, I don't find documentation about CommandHistoryAttribute. Does those need to be configured somewhere or can the software directly set a new attribute without defining anything?

xpromache, 2020-04-02:
Probably we will end up with a list of attributes but we want also the types.

To send a command with attributes:
```
curl -XPOST http://localhost:8090/api/processors/simulator/realtime/commands/YSS/SIMULATOR/SWITCH_VOLTAGE_OFF -d '{
  "sequenceNumber" : 1,
  "origin" : "nico",
  "assignment" : [ {
    "name": "voltage_num",
    "value": "3"
  } ],

  "commandOptions": {
     "attributes" : [{
         "name": "cop1Bypass",
         "value": {
            type: "BOOLEAN",
            booleanValue: true
         }
      }]
   }
}'
```

In your case the value would be something like:
```
 "value": {
            type: "UINT32",
            uint32Value: 34
         }
```

You should see the extra attribute in the command history in yamcs-web.

taguan, 2020-04-02:
Thanks for the detailed explanation, it will take some time for me to put this in place but I can consider this discussion as closed.

closed 2020-04-02 by taguan