yamcs/yamcs #392: Abstract packet name is shown in yamcs-web packet view

taguan, 2020-04-14:
In the Yamcs-web packet view, the displayed name of packets is not the expected one:

[image]

The name that is displayed is the one of the root abstract container. I would have expected the most specific identified container's name to be displayed instead.

xpromache, 2020-04-14:
Looking quickly through the code I think it should show the most specific
identifier - so it might be in your case it doesn't match any sub-packet
from the top.
When you select it, does it show any parameter outside of that root
abstract container?

taguan, 2020-04-14:
I do not see any parameter when clicking on a packet on this view.
However, the parameters that I have defined in my MDB and that are populated from TM packets are correctly assigned to values when looking into the parameters view, so my packets are actually well identified, but there seems to be some mismatch with the packet view.

xpromache, 2020-04-14:
The Packet Viewer decodes the packets internally based on the MDB which can
be either loaded locally or from the server. I presume yours is coming from
the server, right? (when you connect there is an option for that).
If it is the case, then it should produce then the same results like the
server.

Question: did you actually use container inheritance or composition?

We do support container composition but I'm not sure what happens if you do
it at top level - e.g. if you define your container as being  composed from
a header and a body as opposed to our normal way of doing it namely
inheriting from a header container and adding the body parameters in the
child.

taguan, 2020-04-14:
I am not using the offline Packet Viewer but Yamc-web (packet view). So for sure the data is coming from the server, nothing is local.
I am using inheritance, my containers are defined using the specific Yamcs Excel format.

xpromache, 2020-04-14:
Argh, sorry, I thought that your Packet Viewer looks a bit strange...

Where is that view coming from then? Archive?

In the excel there is a flag "a" you can put in the container sheet and
that means to use that container as partition for archiving. Maybe try
setting that on the children if it's not already set. See an example in the
landing.xls in the simulation/mdb directory.

taguan, 2020-04-14:
This is the Packet view below the Telemetry menu.

Adding the "a" flag to child containers did the job! I believe that I didn't find documentation about that flag and that I wrongly interpreted it as meaning"abstract".

thanks!

closed 2020-04-14 by taguan