yamcs/yamcs #396: IntegrationTest.testWsParameterExpiration:467 test failure

leighgarbs, 2020-05-13:
I'm on macOS Mojave 10.14.6 using YAMCS release 4.10.9, but I've seen this in 4.10.8 as well.  There's at least one failure in the tests run by "mvn verify":

```
[INFO] Running org.yamcs.InstancesIntegrationTest
[INFO] Tests run: 2, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 4.325 s - in org.yamcs.InstancesIntegrationTest
[INFO] Running org.yamcs.EventCrashHandlerTest
[INFO] Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.086 s - in org.yamcs.EventCrashHandlerTest
[INFO] 
[INFO] Results:
[INFO] 
[ERROR] Failures: 
[ERROR]   IntegrationTest.testWsParameterExpiration:467
[INFO] 
[ERROR] Tests run: 541, Failures: 1, Errors: 0, Skipped: 14
[INFO] 
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary for Yamcs 4.10.9:
[INFO] 
[INFO] Yamcs .............................................. SUCCESS [  0.169 s]
[INFO] Yamcs :: API ....................................... SUCCESS [ 11.326 s]
[INFO] Yamcs :: Yamcs Client .............................. SUCCESS [  0.305 s]
[INFO] Yamcs :: XTCE ...................................... SUCCESS [  1.183 s]
[INFO] Yamcs :: Core ...................................... FAILURE [04:22 min]
[INFO] Yamcs :: Packet Viewer ............................. SKIPPED
[INFO] Yamcs :: Artemis ................................... SKIPPED
[INFO] Yamcs :: TSE ....................................... SKIPPED
[INFO] Yamcs :: Web Interface ............................. SKIPPED
[INFO] Yamcs :: Distribution .............................. SKIPPED
[INFO] Yamcs :: Example Simulation ........................ SKIPPED
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  04:35 min
[INFO] Finished at: 2020-05-13T11:43:24-07:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-surefire-plugin:3.0.0-M3:test (default-test) on project yamcs-core: There are test failures.
```
I notice that the development setup section of the readme skips the tests, maybe this issue is why it says that.

xpromache, 2020-05-13:
>I notice that the development setup section of the readme skips the tests,
maybe this issue is why it says that.

This is funny :) I think the reason is that it takes a while running the
tests.

We do run the tests after each commit; there is an automated action in
github which runs the tests with java 8 and java 11. I guess it runs on
some sort of Linux.
There was one failure related to alarm un-shelving which appeared
sporadically but I think we nailed it out last week.

The failure you mention (parameter expired not updated) is new to me - is
that very reproducible? What if you run only that test?
mvn -Dtest=IntegrationTest test -pl yamcs-core

leighgarbs, 2020-05-13:
I've tried a full "mvn verify" a couple times and the test failed each time, but then I tried running the test by itself like you suggested and it it only failed once out of three times.  I copied the output I got from the one time the test by itself failed below:
```
[INFO] Running org.yamcs.IntegrationTest
May 13 15:06:42.053 _global [86] org.yamcs.http.RouteHandler [WARNING] 3: Responding '400 Bad Request': SoftwareParameterManager not activated for this processor
May 13 15:06:42.056 _global [86] org.yamcs.http.HttpRequestHandler [WARNING] df31d96a POST /api/processors/IntegrationTest/realtime/parameters:batchSet 400
May 13 15:06:44.165 _global [163] org.yamcs.http.websocket.ParameterResource [WARNING] Received subscribe attempt with 1 invalid parameters. Subscription will continue with 2 remaining valids.
May 13 15:06:50.214 _global [243] org.yamcs.http.RouteHandler [WARNING] 20: Responding '400 Bad Request': Invalid parameters: [name: "/REFMDB/SUBSYS1/InvalidParaName"
]
May 13 15:06:50.214 _global [243] org.yamcs.http.HttpRequestHandler [WARNING] 7c857542 POST /api/processors/IntegrationTest/realtime/parameters:batchGet 400
May 13 15:06:53.231 _global [44] org.yamcs.http.RouteHandler [SEVERE] 22: Blocking the netty thread for 2 seconds. uri: /api/processors/IntegrationTest/realtime/parameters:batchGet
May 13 15:07:02.625 _global [117] org.yamcs.http.RouteHandler [WARNING] 53: Responding '400 Bad Request': Cannot assign aggregate members don't match; expected [member1, member2, member3] but got [member1, member2]
May 13 15:07:02.625 _global [117] org.yamcs.http.HttpRequestHandler [WARNING] 546c84e1 POST /api/processors/IntegrationTest/realtime/parameters:batchSet 400
May 13 15:07:02.636 _global [127] org.yamcs.http.RouteHandler [WARNING] 54: Responding '400 Bad Request': Cannot assign integer from STRING
May 13 15:07:02.636 _global [127] org.yamcs.http.HttpRequestHandler [WARNING] e53180fc POST /api/processors/IntegrationTest/realtime/parameters:batchSet 400
May 13 15:07:04.713 _global [41] org.yamcs.http.websocket.ParameterResource [WARNING] Received subscribe attempt will all-invalid parameters
[ERROR] Tests run: 35, Failures: 1, Errors: 0, Skipped: 1, Time elapsed: 27.064 s <<< FAILURE! - in org.yamcs.IntegrationTest
[ERROR] testWsParameterExpiration(org.yamcs.IntegrationTest)  Time elapsed: 3.023 s  <<< FAILURE!
java.lang.AssertionError
	at org.yamcs.IntegrationTest.testWsParameterExpiration(IntegrationTest.java:469)

[INFO] 
[INFO] Results:
[INFO] 
[ERROR] Failures: 
[ERROR]   IntegrationTest.testWsParameterExpiration:469
[INFO] 
[ERROR] Tests run: 35, Failures: 1, Errors: 0, Skipped: 1
[INFO] 
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  28.926 s
[INFO] Finished at: 2020-05-13T15:07:05-07:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-surefire-plugin:3.0.0-M3:test (default-test) on project yamcs-core: There are test failures.
[ERROR] 
[ERROR] Please refer to /Users/lgarbs/viper/yamcs/yamcs/yamcs-core/target/surefire-reports for the individual test results.
[ERROR] Please refer to dump files (if any exist) [date].dump, [date]-jvmRun[N].dump and [date].dumpstream.
[ERROR] -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoFailureException
```

xpromache, 2020-05-14:
I think I understand what happens. The comment "after 1.5 sec we should get
an set of expired parameters" is wrong. It's true that the packet is
configured with an 1500millisec update interval but there is a tolerance
factor of 1.9 so effectively the expiration time is 1500*1.9=2850millisec.
However it only checks every second for expired parameters so in reality it
might take up to 3850 milliseconds to actually see that the parameter is
expired.
So an easy fix is to change the line 466 from:
pdata = wsListener.parameterDataList.poll(3, TimeUnit.SECONDS);
 into:
pdata = wsListener.parameterDataList.poll(4, TimeUnit.SECONDS);

Can you please try?
I will put probably 4.5 seconds since it takes some time to send the
parameter via websocket and on a slow system hitting a garbage collection
in the middle it might add a few millisec.

thx
nicolae

leighgarbs, 2020-05-14:
I ran just the integration test 20 times with the timeout set to 4 seconds and it still failed a couple times (I lost my record of the test result).

I tried setting the timeout to 4.5 seconds, but that didn't compile, seemingly because it wanted an integer, so I then tried setting the timeout to 5 seconds, and then the test passed 20 successive runs.

Looks like I'm getting further errors when doing a full "mvn verify", even with this timeout adjusted:
```
[WARNING] Tests run: 35, Failures: 0, Errors: 0, Skipped: 1, Time elapsed: 27.047 s - in org.yamcs.IntegrationTest
[INFO] Running org.yamcs.InstancesIntegrationTest
[INFO] Tests run: 2, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 4.352 s - in org.yamcs.InstancesIntegrationTest
[INFO] Running org.yamcs.EventCrashHandlerTest
[INFO] Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.083 s - in org.yamcs.EventCrashHandlerTest
[INFO] 
[INFO] Results:
[INFO] 
[WARNING] Tests run: 535, Failures: 0, Errors: 0, Skipped: 14
[INFO] 
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary for Yamcs 4.10.8:
[INFO] 
[INFO] Yamcs .............................................. SUCCESS [  0.002 s]
[INFO] Yamcs :: API ....................................... SUCCESS [ 11.443 s]
[INFO] Yamcs :: Yamcs Client .............................. SUCCESS [  0.303 s]
[INFO] Yamcs :: XTCE ...................................... SUCCESS [  1.211 s]
[INFO] Yamcs :: Core ...................................... FAILURE [04:21 min]
[INFO] Yamcs :: Packet Viewer ............................. SKIPPED
[INFO] Yamcs :: Artemis ................................... SKIPPED
[INFO] Yamcs :: TSE ....................................... SKIPPED
[INFO] Yamcs :: Web Interface ............................. SKIPPED
[INFO] Yamcs :: Distribution .............................. SKIPPED
[INFO] Yamcs :: Example Simulation ........................ SKIPPED
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  04:34 min
[INFO] Finished at: 2020-05-13T19:46:03-07:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-surefire-plugin:3.0.0-M3:test (default-test) on project yamcs-core: There are test failures.
[ERROR] 
[ERROR] Please refer to /Users/lgarbs/viper/yamcs/yamcs/yamcs-core/target/surefire-reports for the individual test results.
[ERROR] Please refer to dump files (if any exist) [date].dump, [date]-jvmRun[N].dump and [date].dumpstream.
[ERROR] ExecutionException Error occurred in starting fork, check output in log
[ERROR] ExecutionException The forked VM terminated without properly saying goodbye. VM crash or System.exit called?
[ERROR] Command was /bin/sh -c cd /Users/lgarbs/viper/yamcs/yamcs/yamcs-core && /Library/Java/JavaVirtualMachines/jdk1.8.0_251.jdk/Contents/Home/jre/bin/java -jar /Users/lgarbs/viper/yamcs/yamcs/yamcs-core/target/surefire/surefirebooter7501130462964010763.jar /Users/lgarbs/viper/yamcs/yamcs/yamcs-core/target/surefire 2020-05-13T19-41-38_476-jvmRun1 surefire8939304811339837608tmp surefire_976609299061925627343tmp
[ERROR] Error occurred in starting fork, check output in log
[ERROR] Process Exit Code: 134
[ERROR] Crashed tests:
[ERROR] org.yamcs.yarch.rocksdb.BucketDbTest
[ERROR] ExecutionException The forked VM terminated without properly saying goodbye. VM crash or System.exit called?
[ERROR] Command was /bin/sh -c cd /Users/lgarbs/viper/yamcs/yamcs/yamcs-core && /Library/Java/JavaVirtualMachines/jdk1.8.0_251.jdk/Contents/Home/jre/bin/java -jar /Users/lgarbs/viper/yamcs/yamcs/yamcs-core/target/surefire/surefirebooter138257735085215771.jar /Users/lgarbs/viper/yamcs/yamcs/yamcs-core/target/surefire 2020-05-13T19-41-38_476-jvmRun1 surefire486246882063822237tmp surefire_1033906110515685860124tmp
[ERROR] Error occurred in starting fork, check output in log
[ERROR] Process Exit Code: 134
[ERROR] Crashed tests:
[ERROR] org.yamcs.yarch.rocksdb.RdbEngineTest
[ERROR] ExecutionException The forked VM terminated without properly saying goodbye. VM crash or System.exit called?
[ERROR] Command was /bin/sh -c cd /Users/lgarbs/viper/yamcs/yamcs/yamcs-core && /Library/Java/JavaVirtualMachines/jdk1.8.0_251.jdk/Contents/Home/jre/bin/java -jar /Users/lgarbs/viper/yamcs/yamcs/yamcs-core/target/surefire/surefirebooter1004553886960720341.jar /Users/lgarbs/viper/yamcs/yamcs/yamcs-core/target/surefire 2020-05-13T19-41-38_476-jvmRun1 surefire8147462206348238859tmp surefire_1242887265139035212706tmp
[ERROR] Error occurred in starting fork, check output in log
[ERROR] Process Exit Code: 134
[ERROR] org.apache.maven.surefire.booter.SurefireBooterForkException: ExecutionException Error occurred in starting fork, check output in log
[ERROR] ExecutionException The forked VM terminated without properly saying goodbye. VM crash or System.exit called?
[ERROR] Command was /bin/sh -c cd /Users/lgarbs/viper/yamcs/yamcs/yamcs-core && /Library/Java/JavaVirtualMachines/jdk1.8.0_251.jdk/Contents/Home/jre/bin/java -jar /Users/lgarbs/viper/yamcs/yamcs/yamcs-core/target/surefire/surefirebooter7501130462964010763.jar /Users/lgarbs/viper/yamcs/yamcs/yamcs-core/target/surefire 2020-05-13T19-41-38_476-jvmRun1 surefire8939304811339837608tmp surefire_976609299061925627343tmp
[... cut ...]
[ERROR] -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutionException
[ERROR] 
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <args> -rf :yamcs-core
```

xpromache, 2020-05-14:
So at the end it turned out to be a bug in rocksdb appearing in the TestWriter test which was removing a table and the database was left hanging around until another test was picking it up and crashing. It has been fixed here:
https://github.com/facebook/rocksdb/commit/5cd8240b86fd0f0ed07ad4d6ac7cff3130a77ca6#diff-cbb7d5d97c7da536b506a38dd250f053

So I upgraded to latest 5.x branch of RocksDB.

We also discovered that Github CI can run tests on MacOS so I added now Java 8 and 11/MacOS  in the automated tests run executing after each commit:
https://github.com/yamcs/yamcs/actions/runs/104877860

closed 2021-04-20 by fqqb