yamcs/yamcs #398: Support for segmented telemetry SpacePackets using Sequence Flags

AdamElliott-AB, 2020-05-14:
https://public.ccsds.org/Pubs/133x0b1c2.pdf 4.1.2.4.2 describes using the SequenceFlags field in the CCSDS SpacePacket header to indicate that a packet is part of a larger set of user data. If a telemetry packet is segmented this way, does yamcs support assembling the segment-packets together before processing? If not, is this a feature that can be added?

xpromache, 2020-05-14:
Sequence flags from CCSDS packets are not supported; I haven't heard of
anyone using that feature.
What is usually done is multiplexing packets inside CCSDS frames (AOS or
TM) and  Yamcs supports assembling packets across frames.

AdamElliott-AB, 2020-05-15:
Thank you for your response. I appreciate the swift feedback.

closed 2020-05-15 by AdamElliott-AB