yamcs/yamcs #400: Exception received when emitting tuple to subscriber org.yamcs.archive.CcsdsTmIndex

leighgarbs, 2020-05-18:
I'm on macOS 10.14.6 (Mojave) using YAMCS 4.10.9.  When I configure YAMCS with [these configuration files](https://github.com/yamcs/yamcs/files/4647217/config_files.zip) I get the following message when sending YAMCS very small (2-byte) telemetry messages:
```
15:21:10.222 test [26] InternalStream [tm_realtime] Exception received when emitting tuple to subscriber org.yamcs.archive.CcsdsTmIndex@e2b4d7a: java.lang.ArrayIndexOutOfBoundsException: 2
	org.yamcs.utils.ByteArrayUtils.decodeShort(ByteArrayUtils.java:224)
	org.yamcs.tctm.CcsdsPacket.getSequenceCount(CcsdsPacket.java:34)
	org.yamcs.archive.CcsdsTmIndex.onTuple(CcsdsTmIndex.java:108)
	org.yamcs.yarch.Stream.emitTuple(Stream.java:56)
	org.yamcs.management.LinkManager.processTmPacket(LinkManager.java:221)
	org.yamcs.management.LinkManager.lambda$configureDataLink$1(LinkManager.java:162)
	org.yamcs.tctm.UdpTmDataLink.run(UdpTmDataLink.java:69)
	java.lang.Thread.run(Thread.java:748)

15:21:10.222 _global [26] YamcsServer Uncaught exception 'java.lang.ArrayIndexOutOfBoundsException: 2' in thread Thread[Thread-4,5,main]: [org.yamcs.utils.ByteArrayUtils.decodeShort(ByteArrayUtils.java:224), org.yamcs.tctm.CcsdsPacket.getSequenceCount(CcsdsPacket.java:34), org.yamcs.archive.CcsdsTmIndex.onTuple(CcsdsTmIndex.java:108), org.yamcs.yarch.Stream.emitTuple(Stream.java:56), org.yamcs.management.LinkManager.processTmPacket(LinkManager.java:221), org.yamcs.management.LinkManager.lambda$configureDataLink$1(LinkManager.java:162), org.yamcs.tctm.UdpTmDataLink.run(UdpTmDataLink.java:69), java.lang.Thread.run(Thread.java:748)]
15:21:10.222 _global [26] CrashHandler type: UncaughtException, msg: Uncaught exception 'java.lang.ArrayIndexOutOfBoundsException: 2' in thread Thread[Thread-4,5,main]: [org.yamcs.utils.ByteArrayUtils.decodeShort(ByteArrayUtils.java:224), org.yamcs.tctm.CcsdsPacket.getSequenceCount(CcsdsPacket.java:34), org.yamcs.archive.CcsdsTmIndex.onTuple(CcsdsTmIndex.java:108), org.yamcs.yarch.Stream.emitTuple(Stream.java:56), org.yamcs.management.LinkManager.processTmPacket(LinkManager.java:221), org.yamcs.management.LinkManager.lambda$configureDataLink$1(LinkManager.java:162), org.yamcs.tctm.UdpTmDataLink.run(UdpTmDataLink.java:69), java.lang.Thread.run(Thread.java:748)]
```
I use "netcat" to reproduce the problem as shown here, once YAMCS is up and running:
```
echo -n ab | nc -u localhost 18202
```
This command sends a 2-byte UDP message to the port where YAMCS should be configured to be listening.  The "test.xml" XTCE file defines a SequenceContainer with two 1-byte unsigned integer fields.  Looking at the YAMCS server GUI "Packets" page after sending the message does show the 2-byte packet received with the right data, but the "Parameters" page doesn't show the decoded packet data (it shows "-" in the Value column for both parameters).

On a possibly related note, increasing the number of fields in the "test.xml" SequenceContainer to 5 1-byte unsigned integer fields and then sending a 4-byte UDP message with this command ...
```
echo -n abcd | nc -u localhost 18202
```
... produces this output ...
```
15:43:07.363 test [26] SequenceContainerProcessor Got java.lang.ArrayIndexOutOfBoundsException when extracting from the buffer of length 32 bits, bitPosition 32 entry: ParameterEntry position:4, container:TestPacket locationInContainer:0 from:previousEntry, ParaName: Field5 paraType:IntegerDataType name:UINT8 sizeInBits:8 signed:false encoding:IntegerDataEncoding[sizeInBits: 8, byteOrder: BIG_ENDIAN, encoding:UNSIGNED] aliases: {}
```
... but unlike the first example, the data in the received message is decoded and displayed as values for the first four XTCE fields in the "Parameters" page in the GUI.  No value is shown for the fifth field, which makes sense since the message has only enough data for the first four fields.

xpromache, 2020-05-19:
1. You seem to have hit a relic from when Yamcs was only allowing CCSDS
packets (7 bytes min). The IndexServer service loads a CcsdsIndex which
produces the completeness index which you can see in the Web interface when
you go to Archive -> Overview. But that view shows other type of data
called histogram data which is build differently than the CcsdsIndex. The
index server is the one producing that view, you can see it in the config
file:
 - class: org.yamcs.archive.IndexServer
   args:
     streams:
       - tm_realtime
       - tm_dump

To get rid of the CcsdsIndex you can comment out the whole service but then
you lose not only the CCSDS completeness index but also the histograms.
To get along with small packets so they are not sent to the CcsdsIndex, you
can remove just tm_realtime from the list.

The whole config is confusing because no way someone can know those two
streams get passed to the CcsdsIndex and also it's not possible to disable
the CcsdsIndex alone unless you have your own index as a replacement. To be
fixed.

For now, if you want small packets on a stream, please comment it out from
there (unfortunately you cannot leave the list empty so you may need to
define a bogus stream to have at least one entry). If your packets are not
CCSDS you can comment it out anyway since it will produce garbage.

2. The error is indeed if you defined parameters that are beyond the limit
of the binary packet. There is an option to silence it if you want but I
guess it's better to leave it in during development when you want to get
your Mission Database correct. I'm going to change the error message to
make it less suspicious that something has crashed.

leighgarbs, 2020-05-19:
Thanks for looking at this.  I can see that removing the IndexServer service from my config does what you say (it stops the crashing and also effectively makes the Archive -> Overview page stop working).  Fortunately we expect all our telemetry to be CCSDS space packets, at least for the forseeable future, so I doubt we'll see much of this in practice.

I imagine we'll upgrade to a YAMCS release with a fix for this (when one is available) so we don't have people sending YAMCS strange telemetry and bringing YAMCS down unintentionally.

xpromache, 2020-05-19:
I fixed now the code that the CcsdsIndex will ignore the short packets.

I think the pre-processor should not let those short packets pass through,
which pre-processor are you using?

leighgarbs, 2020-05-19:
> which pre-processor are you using?

I'm using the GenericPacketPreprocessor, configured to not read the timestamp or sequence count from within the packet (-1 for both configuration arguments).

> I think the pre-processor should not let those short packets pass through

That's a good point.  I used the GenericPacketPreprocessor to demonstrate this issue because it was convenient, but the preprocessors we use for production (probably not the GenericPacketPreprocessor) will do this sort of filtering, at least to the extent they can.

closed 2020-06-07 by xpromache