yamcs/yamcs #403: Command significance levels are not inherited from the BaseCommand

merose, 2020-05-24:
According to the current XTCE schema, when defining a `<DefaultSignificance>` inside a `<MetaCommand>`, the significance level is supposed to be inherited from the `<BaseCommand>`, if there is a base command specified. The documentation string for the `<DefaultSignificance>` element says this:

"Some Command and Control Systems may require special user access or confirmations before transmitting commands with certain levels.  The level is inherited from the Base MetaCommand."

However, Yamcs does not inherit the default significance from the base command, according to the display in the MDB area in the Yamcs web UI.

xpromache, 2021-04-21:
this should work now on the server, yamcs-web still to be adapted.

fqqb, 2021-05-03:
On master

closed 2021-05-03 by fqqb