yamcs/yamcs #405: Yamcs should support <ValidRange> for parameter types and <ValidRangeSet> for argument types

merose, 2020-05-27:
Yamcs supports loading valid ranges from XSL files, but there is currently no way to set those ranges from within an XTCE file.

merose, 2020-05-27:
Actually, since there are Alarms, the valid range in a parameter type is not super important. But some way of validating a command argument would be highly useful.

xpromache, 2020-05-27:
Command argument valid range is supported for commands but not read from XML.
This can be implemented quickly. To enable some compatibility with XTCE 1.2 I will add reading of ValidRangeSet with only one range (more will throw an exception for now).

Do you have a sample XTCE I can use in unit tests?

merose, 2020-05-29:
Here's a sample XTCE which has both a parameter and an argument with a validity range. GitHub required me to add a ".txt" extension. The correct name is "example-xtce.xml".

[example-xtce.xml](https://github.com/yamcs/yamcs/files/4704587/example-xtce.xml.txt)

xpromache, 2021-04-21:
Closing for now as ValidRangeSet with one range is supported now.

Multiple ranges are not supported, can be added later if anybody needs them.

closed 2021-04-21 by xpromache