yamcs/yamcs #406: Would be useful to support "baseType" attribute for parameter and argument types

merose, 2020-05-28:
It is useful in XTCE to use the _baseType_ attribute to reduce the complexity of parameter and argument type definitions. For example, one could define a basic floating point parameter type and then specialize the type by just adding units and alarms.

      <FloatParameterType name="float64_msb" sizeInBits="64">
        <FloatDataEncoding sizeInBits="64" byteOrder="mostSignificantByteFirst" encoding="IEEE754_1985" />
      </FloatParameterType>
      <FloatParameterType name="wind_direction_t" baseType="float64_msb">
        <UnitSet>
          <Unit>degrees</Unit>
        </UnitSet>
      </FloatParameterType>

Currently Yamcs does not examine the _baseType_ attribute. It seems like it would be fairly easy to add. Consider lines 850-851 of XtceStaxReader.readFloatParameterType():

        String value = readMandatoryAttribute("name", element);
        floatParamType = new FloatParameterType(value);

Perhaps changing to something like this:

        String value = readMandatoryAttribute("name", element);
        String baseTypeName = readAttribute("baseType", element, null);
        if (baseTypeName == null) {
            floatParamType = new FloatParameterType(value);
        } else {
            baseType = spaceSystem.getParameterType(baseTypeName);
            if (baseType == null || !(baseType instanceOf FloatParameterType)) {
                ... handle error ...
            }
            floatParamType = (FloatParameterType) baseType.copy();
            floatParamType.setName(value);
        }

There may be some refactoring that could make the code less lengthy. For example, adding a copy constructor that takes a new name and a base type, which may be null:

        FloatParameterType(String name, FloatParameterType baseType) { ... }

and maybe a generic method in XtceStaxReader for getting the base type, something like this:

        public <T extends ParameterType> T getBaseType(StartElement element, SpaceSystem spaceSystem, Class<T> clazz) {
            String baseTypeName = readAttribute("baseType", element, null);
            if (baseTypeName == null) {
                return null;
            } else {
                ParameterType baseType = spaceSystem.getParameterType(baseTypeName)
                if (baseType == null) {
                    ... handle error ...
                } else if (!clazz.isAssignableFrom(baseType.getClass())) {
                    ... handle error ...
                } else {
                    return clazz.cast(baseType);
                }
            }
        }

Then the original two lines of code in `XtceStaxReader.readFloatParameterType()` become:

        String value = readMandatoryAttribute("name", element);
        floatParamType = new FloatParameterType(value, getBaseType(element, spaceSystem, FloatParameterType.class));

which isn't so onerous, since it must be duplicated for each parameter type and argument type possibility. (You need a second generic method for getting the base type for  arguments, though.)

xpromache, 2020-05-29:
Thanks for your suggestion. One issue I see is that it can reference types not yet known at the moment of parsing, for example from other files.

We have a class called UnresolvedNameReference which can be used to add later characteristics to an object but I'm not sure it works with constructors.
I shall give it a try.

First thing to do is to create a reference yamcs XTCE file set similar with the refmdb.xls we use for testing the XLS parser. So far we have avoided this by using other's people XML files like the BogusSat I copied from the xtcetools project.

merose, 2020-05-29:
I forgot about forward references, or references to other files. Yes, that complicates things. For what it's worth, here's an example XTCE file that uses `baseType`. It uses forward references for both parameter and argument types. The references are to a nested SpaceSystem in the same XTCE. I like your suggested approach of creating a comprehensive XTCE mission database for testing, similar to refmdb.xls, but I understand that's a lot of work.

Again, GitHub requires a .txt extension. The correct file name is "example-xtce.xml".

[example-xtce.xml](https://github.com/yamcs/yamcs/files/4704617/example-xtce.xml.txt)

closed 2020-06-12 by xpromache