yamcs/yamcs #409: Yamcs Telemetry web UI does not display AbsoluteTimeParameter values occurring on leap seconds

merose, 2020-06-01:
To reproduce:
1. Load an XTCE mission database containing an AbsoluteTimeParameter in a SequenceContainer.
2. Send a packet with a time value occurring on a leap second.
3. Go to the Telemetry tab in the Yamcs web UI, navigate to the time parameter, and view Historical Data.

Expected result: The parameter engineering value shown should have a time of 23:59:60.
Actual result: The parameter value engineering value is blank.

In my case the packet generation time is obtained from this time parameter. That column is blank for the row that should show the leap second.

[image]

This is the AbsoluteTimeParameterType definition:

      <AbsoluteTimeParameterType name="PacketTimeType"
        shortDescription="The time format used for the packet generation time">
        <LongDescription>
          The time format used to encode the packet generation time for the CCSDS
          secondary header. We use a 48-bit format, where the first 32 bits are the
          integral number of seconds since the UNIX epoch (1970-01-01T00:00:00Z),
          and the last 16 bits are the fractional seconds in units of 1/65536 seconds.
          The times are in TAI seconds, so we have to offset by -9 seconds, since
          Yamcs assumes that a UNIX epoch means the times are UTC seconds, and the
          offset from UTC to TAI was 9 seconds at the Unix epoch. 
        </LongDescription>
        <Encoding units="seconds" scale=".0000152587890625" offset="-9">
          <IntegerDataEncoding sizeInBits="48" byteOrder="mostSignificantByteFirst" />
        </Encoding>
        <ReferenceTime>
          <Epoch>UNIX</Epoch>
        </ReferenceTime>
      </AbsoluteTimeParameterType>

Note that the HTTP API is returning the proper UTC time string:

    $ curl http://localhost:8090/api/archive/myproject/paraeters/Demo/ccsdsPacketTime?limit=1
    {
      "parameter": [{
        "id": {
          "name": "ccsdsPacketTime",
          "namespace": "/Demo"
        },
        "rawValue": {
          "type": "UINT64",
          "uint64Value": "94090614210560"
        },
        "engValue": {
          "type": "TIMESTAMP",
          "stringValue": "2015-06-30T23:59:60.000Z",
          "timestampValue": "1435708835000"
        },
        "generationTime": "1435708835000",
        "acquisitionStatus": "ACQUIRED",
        "processingStatus": true,
        "generationTimeUTC": "2015-06-30T23:59:60.000Z"
      }],
      "continuationToken": "eyJ0aW1lIjoxNDM1NzA4ODM1MDAwfQ"
    }

xpromache, 2020-06-02:
Thanks for the report, it uncovers (yet again...) some unfinished business in Yamcs.

Once upon a time when we only had a java client we were sending the internal Yamcs time (milliseconds since 1970 with leap seconds added).
This corresponds to the "generationTime": "1435708835000" you see in the response. 
If you would convert this straight to UTC using a Unix time conversion you will get 2015-07-01T00:00:35 so 35 leap seconds in front of UTC.

Sending this time had the disadvantage that the clients need the leap second table to be able to convert the time to UTC.

When we decided to be more open to other clients (which don't have leap second tables at hand), we tried to solve this problem by introducing the additional parameter generationTimeUTC which contains a string formatted UTC time (the formatting is done on the server by the TimeEncoding class). 

Later on it seemed silly to have two timestamps sent all the time and we also discovered how google handles it with their timestamps used also in protobuf. They "smear" the time around the leap seconds:
https://developers.google.com/time/smear

 So we thought the protobuf approach is the best of the two worlds because:
1. No need for a leap second table, if you don't care about leap seconds you get slightly off timestamps around the leap seconds.
2. If you do care about leap seconds (and you have a leap second table) you can "unsmear" the timestamp on the client. The protobuf timestamp having nanosecond resolution and the original having only millisecond resolution, you can recover the exact original millisecond timestamp.

We have converted most of the API to the protobuf timestamp but inexplicably the parameter value didn't get converted. But now that you bring it up I think we should do it before the next version.
Practically the "generationTime" will be sent as a protobuf Timestamp which maps to a smeared UTC string when formatted to json (in binary protobuf it is mapped as a struct second/nanosecond).

We will leave the generationTimeUTC  a little longer, the clients that use that string will continue to work.

And yes, we should also fix the web UI in the process. Probably this should do the right thing and undo the smearing using the leap table it gets from the server also via the API.

I think most of the people do not care about leap seconds and before this report, the only time somebody complained about it was a person saying that a 23:59:60 timestamp was wrong (although it was correct). He was processing some data extracted from Yamcs and his scripts were choking in that timestamp. He "solved" the problem by just ignoring the data point...

merose, 2020-06-02:
In our OpenMCT plugin we've seen performance issues when using the string format returned in generationTimeUTC. That is, the parsing time to convert back to a Date – which is needed internally – is seen as a bottleneck by the OpenMCT team. However, we understand that we need to convert from TAI seconds to UTC seconds, possibly using the leap second table. Getting the "smeared" UTC seconds as well as the TAI seconds would give us two options. I'm not yet sure which we'd pick.

merose, 2020-06-02:
I've asked the OpenMCT team for their input on how to handle the TAI/UTC time difference in OpenMCT. I'm a little concerned about using the smeared time, since it won't correspond to the system clock (which also might be smeared, in a different way, depending on how the system administrators apply the leap second correction to the OS clocks). I'll have to engage our system administrators to see how they usually handle leap seconds. They probably just let the machines adjust based on the NTP server, but I'm not sure over what interval that usually occurs.

xpromache, 2020-06-02:
> In our OpenMCT plugin we've seen performance issues when using the string format returned in generationTimeUTC. That is, the parsing time to convert back to a Date – which is needed internally – is [...]

I'm surprised parsing the timestamp is posing problems; I thought the reason protobuf serializes it that way is for javascript to parse it easily (but then I don't know much about javascript). 

In fact we want to get rid of the TAI seconds because as I said if one has the leap seconds table, it can get back to the TAI from the smeared time. We have an API call that returns the leap seconds table used by the server, it's also visible in the web interface.

> I'm a little concerned about using the smeared time, since it won't correspond to the system clock (which also might be smeared, in a different way, depending on how the system administrators apply the leap second correction to the OS clocks).

I don't really understand which system you are talking here about - the ground system or the flight system? 
If you are talking about the ground systems, I think we do not handle correctly in Yamcs. We use System.currentTimeMilliseconds() from Java but although in principle we have the leap seconds table, we have no information about the smearing process that might be in effect on the local computer. Something to think about...

akhenry, 2020-07-10:
> I'm surprised parsing the timestamp is posing problems

I can understand why that would be surprising. It's because of how we have abstracted time handling in Open MCT. Our API supports non terrestrial time systems that can't be represented by JavaScript native dates, and as such as long ago we decided on the convention of just representing all times as numbers, and then defining "time systems" that know how to make sense of those numbers. We used numbers because we need to be able to do greater-than and less-than comparisons at very high data rates in a web browser.

Consequently, all telemetry gets converted from its wire format to a number. For display, it then gets converted to a context-specific human readable string (some displays may want to represent the timestamp differently, eg. zoom-sensitive ticks on a plot, or the user may apply custom formatting). Therefore, it's most performant if we use numbers for the wire format itself, as this effectively avoids the step of first parsing a string to a number for every single telemetry value we receive.

akhenry, 2020-07-17:
It's also worth mentioning that creating JavaScript native dates from strings using the builtin functions (either `.parse()` or the `Date` object constructor) is [discouraged](https://developer.mozilla.org/it/docs/Web/JavaScript/Reference/Global_Objects/Date/parse) due to inconsistencies in implementations. Instead developers are encouraged to use third party libraries (we use [Moment.js](https://momentjs.com/) for this). Doing this parsing in non-native code, with hundreds or thousands of values per second, is why we see performance issues.

So, in my opinion, the optimal format for JavaScript users to receive timestamps would be
* In ms since the UNIX epoch, ignoring leap seconds. My feeling is that smearing would probably work ok as well, but I'm not 100% sure. If I'm understanding it correctly, the differences between the OS clock and the YAMCS timestamps will be small, and would happen very rarely (just in the 24 hours preceding a leap second event).  I'm not sure that there would be any serious consequences to the user. It would certainly be more performant than dealing with a leap second table.
* As a number (and not a string representation of a number. ie. `generationTime: 1234` not `generationTime: '1234'` although to be honest this doesn't make a *huge* difference),

fqqb, 2020-07-17:
We are using Google’s protobuf library for generating JSON messages and as per https://github.com/protocolbuffers/protobuf/blob/master/src/google/protobuf/timestamp.proto the JSON-serialization of Timestamps follows a very specific format (RFC 3399 - similar but more strict than ISO8601) and even only emits UTC timestamps (Z suffix). To my best knowledge this should not cause any browser parsing issues. I think such issues are more related to the use of unconventional date formats or the use of timezones.

Around the period of a leap second insertion, this format will show more accuracy than milliseconds (up to nanoseconds), which is information that you could not represent with just a milliseconds counter.

If performance is a concern, my suggestion would be to use `string -> Date` native browser parsing functions rather than any high-level parsing. And if you require the use of moment for other reasons, then you could do `Date -> Moment`, which should bypass the slower moment parsing functions (I think Moment just wraps a Date object).

As an alternative solution, you may also want to access the API without JSON, and just use the native Protobuf binary format (using HTTP header `Accept: application/protobuf`). The protoc binary has nowadays an option available for generating the necessary JavaScript routines: https://developers.google.com/protocol-buffers/docs/reference/javascript-generated
I have not experimented with it myself yet, so I don’t know how well it otherwise stacks up against JSON parsing. I also don’t know how it handles Timestamp messages, although I assume it would provide direct access to the seconds and nanos integers.

> As a number (and not a string representation of a number. ie. generationTime: 1234 not generationTime: '1234' although to be honest this doesn't make a huge difference),

Specific to this point: JSON-serialization of integers who’s datatype exceed the JSON specification is automatically done with strings. You can find some more information about the reasons for that here: https://developers.google.com/discovery/v1/type-format. This happens in a number of locations throughout our APIs (basically wherever we use uint64 or int64).