yamcs/yamcs #415: There should be a configuration parameter to specify the HttpServer bind address

merose, 2020-06-13:
Currently there is a configuration parameter to HttpServer to specify the port Yamcs should respond to for web requests. However there is no parameter to specify the address to which the server should be bound. Instead, in _HttpServer.createAndBindBootstrap()_ a wildcard address is used:

        bootstrap.bind(new InetSocketAddress(port)).sync();

There should be a configuration parameter to specify the address to bind to. If not specified, the above code should be used. If specified, this code should be used instead:

        bootstrap.bind(new InetSocketAddress(address, port)).sync();

where the `address` parameter is a String. For example, if the address parameter were configured as `127.0.0.1`, then Yamcs would only respond to requests from localhost.

This is also important when run on a multi-homed server.

fqqb, 2020-06-18:
Fixed on master by adding an `address` configuration property.

In current implementation this property is shared for both HTTP and HTTPS. In future work, I'm thinking to add an arbitrary amount of *connectors* each with their own properties, but this work will likely be timed together with an other idea we have, which is to make the HttpServer a standard singleton service of Yamcs (i.e. without requiring to be configured in global services).

closed 2020-06-18 by fqqb

0x4ndy, 2025-02-26:
> Fixed on master by adding an `address` configuration property.

How do I configure it? Can you provide an example of this config file and its location?

The issue I have now is that `yamcs` starts on what the host name is:
```
...
08:22:23.013 sle [1] YamcsServerInstance Awaiting start of service CommandHistoryRecorder
08:22:23.013 sle [1] YamcsServerInstance Awaiting start of service ParameterArchive
08:22:23.020 _global [1] YamcsServer Yamcs started in 4041ms. Started 1 of 1 instances and 11 services
08:22:23.024 _global [1] WebPlugin Website deployed at http://localhost.localdomain:8090
^C08:22:30.193 _global [19] YamcsServer Yamcs is shutting down
08:22:30.210 sle [19] LinkManager Awaiting termination of link SLE_OUT.vc0
08:22:30.211 sle [19] LinkManager Awaiting termination of link SLE_OUT
...
```

And I want it to start on 0.0.0.0 or a specific IP of one of my NICs.

fqqb, 2025-02-26:
0.0.0.0 is default behavior.

You can configure it explicitly in the options of your HttpServer service in etc/yamcs.yaml.
See https://docs.yamcs.org/yamcs-server-manual/services/global/http-server/

Sth like this:

```yaml
services:
  - class: org.yamcs.http.HttpServer
    args:
      port: 8090
      address: 0.0.0.0
```

But I expect even without this configuration, you should find it to work. The log statement just prints any matching host name.

0x4ndy, 2025-02-26:
That worked, thanks!