yamcs/yamcs #419: Enable platform detection in Maven for rocksdb compile to enable aarch64 / Raspberry Pi Support for YAMCS Server

nmaas87, 2020-06-23:
Dear all,

I have been trying to use yamcs-server on Raspberry Pi for some time, however, it initially failed due to the dependency protobuffers not been supported on ARM 32 bit platfoms, but only on ARM v64 ( https://github.com/protocolbuffers/protobuf/releases/tag/v3.7.1 ).
However, since the RPi now got an 64 bit Beta Image ( https://www.raspberrypi.org/forums/viewtopic.php?t=275370 ), I thought I might try again - as the RPi now identifies as aarch64 (ARM 64 bit) platform (Linux raspberrypi 5.4.42-v8+ #1319 SMP PREEMPT Wed May 20 14:18:56 BST 2020 aarch64 GNU/Linux).

So I went ahead to compile yamcs on a fresh install of said platform, on an RPi 3B+ (I got a RPi 4 around here as well, but as yamcs is quite moddest in its use of resources (thanks for the good work!) - that is not really needed :)). Basically just a basic install with AdoptOpenJDK 8. 

````
wget -qO - https://adoptopenjdk.jfrog.io/adoptopenjdk/api/gpg/key/public | sudo apt-key add -
sudo apt update
sudo apt install -y software-properties-common
sudo add-apt-repository --yes https://adoptopenjdk.jfrog.io/adoptopenjdk/deb/
sudo apt update
sudo apt install adoptopenjdk-8-hotspot maven zip npm
wget https://github.com/yamcs/yamcs/archive/yamcs-4.10.9.zip
unzip yamcs-4.10.9.zip
cd yamcs-yamcs-4.10.9
mvn clean install -DskipTests
cd yamcs-web/src/main/webapp
npm install
npm run build
cd -
./run-simulation.sh
````

The good news first: It does scuccessfully build the jars. Maven is even intelligent enough to pull the aarch64 vesion of protobuffers - which is awesome, as the compile failed in "the old times" at that point. Starting the simulation / yamcs however does not work, it seems there is one last piece missing, which is the hard-compile x64 version of the RocksDB:

````
pi@raspberrypi:~/yamcs-yamcs-4.10.9 $ ./run-simulation.sh
[INFO] Scanning for projects...
[INFO]
[INFO] ------------------------< org.yamcs:simulation >------------------------
[INFO] Building Yamcs :: Example Simulation 4.10.9
[INFO] --------------------------------[ jar ]---------------------------------
[INFO]
[INFO] >>> yamcs-maven-plugin:1.2.1:run (default-cli) > process-classes @ simulation >>>
[INFO]
[INFO] --- maven-resources-plugin:3.1.0:resources (default-resources) @ simulation ---
[INFO] Using 'UTF-8' encoding to copy filtered resources.
[INFO] Copying 7 resources
[INFO]
[INFO] --- maven-compiler-plugin:3.8.0:compile (default-compile) @ simulation ---
[INFO] Nothing to compile - all classes are up to date
[INFO]
[INFO] <<< yamcs-maven-plugin:1.2.1:run (default-cli) < process-classes @ simulation <<<
[INFO]
[INFO]
[INFO] --- yamcs-maven-plugin:1.2.1:run (default-cli) @ simulation ---
[INFO] Creating configuration at /home/pi/yamcs-yamcs-4.10.9/simulation/target/yamcs
Exception in thread "main" 07:22:24.249 _global [1] stdout java.lang.UnsatisfiedLinkError: /tmp/librocksdbjni8972936723257032940.so: /tmp/librocksdbjni8972936723257032940.so: cannot open shared object file: No such file or directory (Possible cause: can't load AMD 64-bit .so on a AARCH64-bit platform)
07:22:24.252 _global [1] stdout         at java.lang.ClassLoader$NativeLibrary.load(Native Method)
07:22:24.253 _global [1] stdout         [ t java.lang.ClassLoader.loadLibrary0(ClassLoader.java:1;31934)
407:m2I2:24.25N3 F_O[mlob] al1m1] stdout        at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1817)
07:22:24.254 _global [1] stdout         at java.lang.Runtime.load0(Runtime.java:809)
07:22:24.255 _global [1] stdout         at java.lang.System.load(System.java:1086)
07:22:24.255 _global [1] stdout         at org.rocksdb.NativeLibraryLoader.loadLibraryFromJar(NativeLibraryLoader.java:78)
07:22:24.256 _global [1] stdout
[INFO] BUILD SUCCESS
[INFO] ------------------------------------------------------------------------0m[ 0;33m        at org.rocksdb.NativeLibraryLoader.loadLibrary(Nativ
eLibraryLoader.java:56)
07:22:24.257 _global [1] stdout         at org.rocksdb.RocksDB.loadLibrary(RocksDB.java:64)
07:22:24.257 _global [1] stdout         at org.rocksdb.RocksDB.<cli[INFO] Total time:  13.891 s
[ni>([R1o;cksD34mINFO] Finished at: 2020-06-23T07:22:24+01:00
[INFO] ------------------------------------------------------------------------
B.java:35)
07:22:24.258 _global [1] stdout         at org.rocksdb.FlushOptions.<clinit>(FlushOptions.java:9)
07:22:24.258 _global [1] stdout         at org.yamcs.yarch.rocksdb.RDBFactory.<clinit>(RDBFactory.java:42)
07:22:24.259 _global [1] stdout         at org.yamcs.YamcsServer.prepareStart(YamcsServer.java:1020)
07:22:24.259 _global [1] stdout         at org.yamcs.YamcsServer.main(YamcsServer.java:827)
````

However, it seems like there is now also aarch64 support for the current rocksDB Versions:
https://github.com/facebook/rocksdb/issues/5559  /   https://github.com/facebook/rocksdb/wiki/Platform-Requirements
Probably it just would need a recompile or some magic to Maven to compile rocksDB according to the platform (i.e. x64 or aarch64) to make YAMCS working on all kinds of ARMv64 servers and computers.

I think it would be quite a cool feature, not only showcasing the optimization and performance of YAMCS in contrast to different Mission Control Systems (especially legacy ones from certrain big companies, which you cannot use without a whole VMware Cluster setup) - as a test system - but also in lab use. And, I admit, due to use of COTS components, you are now able to build very small and capable payloads - which can be easily interfaced for testing and checkouts with an Raspberry Pi or similar. Running the YAMCS Server also on that platform would be an awesome feature to make such checkouts easier.

Thanks a lot :)

nmaas87, 2020-06-23:
Added info to the pull request for the aarch64 support of the used RocksDB in YAMCS:
https://github.com/facebook/rocksdb/pull/6250

xpromache, 2020-06-23:
> facebook/rocksdb#6250 <https://github.com/facebook/rocksdb/pull/6250>

Hi Nico,

We have recently updated the rocksdb version and the new version has a arm
build inside, so it might work out of the box in the next version.
I also got recently a raspbery 4 to try it out but didn't have the time yet.

nicolae

nmaas87, 2020-06-24:
I tried building the rocksDB myself on the RPi 3, but after it gaining over 28 GB on the SDcard, I had to abort. Also it did not like the RPi 3 as it really started to swap, so on a RPi 4 with 4 GB, the compile worked better, but still I had the error of the full sd card  - so I decided to ditch that try and just git clone the current yamcs repo and build that:

````
[INFO] --- os-maven-plugin:1.6.2:detect (default) @ yamcs-api ---
[INFO] ------------------------------------------------------------------------
[INFO] Detecting the operating system and CPU architecture
[INFO] ------------------------------------------------------------------------
[INFO] os.detected.name: linux
[INFO] os.detected.arch: aarch_64
[INFO] os.detected.version: 5.4
[INFO] os.detected.version.major: 5
[INFO] os.detected.version.minor: 4
[INFO] os.detected.release: debian
[INFO] os.detected.release.version: 10
[INFO] os.detected.release.like.debian: true
[INFO] os.detected.classifier: linux-aarch_64
````

It detected the aarch_64 correctly but went on to grab several versions of rocksdb, none of those I thought would work:

````
 --- maven-install-plugin:2.4:install (default-install) @ yamcs-web ---
[INFO] Installing /home/pi/yamcs/yamcs-web/target/yamcs-web-5.0.0-SNAPSHOT.jar to /home/pi/.m2/repository/org/yamcs/yamcs-web/5.0.0-SNAPSHOT/yamcs-web-5.0.0-SNAPSHOT.jar
[INFO] Installing /home/pi/yamcs/yamcs-web/pom.xml to /home/pi/.m2/repository/org/yamcs/yamcs-web/5.0.0-SNAPSHOT/yamcs-web-5.0.0-SNAPSHOT.pom
[INFO]
[INFO] -----------------------< org.yamcs:distribution >-----------------------
[INFO] Building Yamcs :: Distribution 5.0.0-SNAPSHOT                     [9/14]
[INFO] --------------------------------[ pom ]---------------------------------
Downloading from central: https://repo.maven.apache.org/maven2/org/rocksdb/rocksdbjni/5.18.4/rocksdbjni-5.18.4-linux64.jar
Downloading from central: https://repo.maven.apache.org/maven2/org/rocksdb/rocksdbjni/5.18.4/rocksdbjni-5.18.4-osx.jar
Downloading from central: https://repo.maven.apache.org/maven2/org/rocksdb/rocksdbjni/5.18.4/rocksdbjni-5.18.4-win64.jar
Downloaded from central: https://repo.maven.apache.org/maven2/org/rocksdb/rocksdbjni/5.18.4/rocksdbjni-5.18.4-win64.jar (2.1 MB at 1.4 MB/s)
Downloaded from central: https://repo.maven.apache.org/maven2/org/rocksdb/rocksdbjni/5.18.4/rocksdbjni-5.18.4-osx.jar (2.5 MB at 1.5 MB/s)
Downloaded from central: https://repo.maven.apache.org/maven2/org/rocksdb/rocksdbjni/5.18.4/rocksdbjni-5.18.4-linux64.jar (8.7 MB at 3.9 MB/s)
[INFO]
````

However, running the simulator with the run_example.sh simulator - it DOES work. Actually very well! I could see it running with a systemload of about 0.4 (next to nothing) and consuming under 800 MB RAM (this is with active Raspberry Pi OS Desktop, as I did not disable it) - thats quite impressive. I saw some errors, i.e. that yamcs crashed as I tried to open the first OPI in the web interface - the rest seemed to work and it did also recover... So, YAMCS Server 5 seems to be able to work on ARM64 just fine!

nmaas87, 2020-06-24:
Just one question left: Are you planing on packing an aarch64 release, will the normal release also support aarch64 out of the box (probably not possible I guess), or do we need to compile the releases ourself? (I guess the last one :))

fqqb, 2020-06-24:
Congrats on being the first to make this work. Some explanations for what you are observing:

> It detected the aarch_64 correctly but went on to grab several versions of rocksdb, none of those I thought would work:

That's because the distribution pom.xml makes binary tarballs for confirmed-working architectures (currently win64, linux64 and osx). These builds include only the JNI libs specific to each architecture.

> However, running the simulator with the run_example.sh simulator - it DOES work

And that's because our development pipeline uses a (fat) rocksdbjni dependency which includes all available JNI libs (previously aarch64 was not part of that).

In the distribution builds we try to be more specific, because the size difference is very significant.

>  I saw some errors, i.e. that yamcs crashed as I tried to open the first OPI in the web interface

It's been like that for a while. We have just begun to start adding support for OPI files in the web interface. It's a lot of work because there are so many widgets with so many properties, and we basically need to duplicate the entire implementation. The OPI files in the simulation example are not yet working because of some advanced widgets like tabs and others. It's on our neverending TODO list.

> Are you planing on packing an aarch64 release

Yes, once @xpromache also has this running successfully on his device, we'll add this to our generated binary builds. The upgrade to a later rocksdbjni version was long due, it also allows us to now run on Windows without any issue.

fqqb, 2020-06-24:
> I saw some errors, i.e. that yamcs crashed as I tried to open the first OPI in the web interface

Actually, it's expected that the display doesn't fully work, but it shouldn't crash Yamcs. Maybe a bug.

nmaas87, 2020-06-25:
Thanks for your awesome work!
- Nice to hear that the "Web Displays" are being worked on. Does this also include support for the "old" USS Displays in i.e. Web or YAMCS Client?
- Yeah, I saw the Windows build in my self-compiled folders and while I actually prefer to run YAMCS on Windows in Microsofts WSL / Windows Subsystem for Linux (which has been working perfectly for a long time :)) - I guess it makes sense for some users :)
- The aarch64 release is very welcome, especially for use with testing equipment in a lab situation and on smaller missions
- I cannot say if this OPI did really "crash" YAMCS: It definitly lost connection to the web frontend, so I guess only the node part died - the YAMCS Server component seemed to be still working and it kinda "repaired itself" within seconds, so after some refreshes in the webbrowser, it was "online" again. So its maybe not yet perfect (which I don't expect here) - but it still is far away from some really devastating crash - so quite fine now :)

fqqb, 2020-06-25:
> Does this also include support for the "old" USS Displays in i.e. Web or YAMCS Client?

Due to this not being generically useful to most people, it's been moved to an internal repo some time ago. I wouldn't mind repackaging this as a plugin to Yamcs, but it's technically quite challenging (not easy to make dynamically linked plugins in an angular website), so it's not something we're actively pursuing right now.

xpromache, 2020-07-29:
yamcs 5.0.0 has been (minimally) tested on a raspberry pi 4,  the getting started instructions or the examples work unmodified.
closing.

closed 2020-07-29 by xpromache