yamcs/yamcs #428: Cannot create service XtceTmRecorder, with arguments {}: null

leighgarbs, 2020-07-29:
I'm attempting to run Yamcs 5.0.0 with a configuration that works in Yamcs 4.10.9 and uses the "subLoaders" key in the mdb section of the instance configuration to load a couple XTCE files within the context of a top-level one.  It looks something like this:
```yaml
  - type: xtce
    args:
      file: xtce1.xml
      
  - type: xtce
    args:
      file: xtce2.xml
      
    subLoaders:
      - type: xtce
        args:
          file: xtce3.xml
```
It looks like this no longer works in Yamcs 5.0.0 because when I run Yamcs 5.0.0 using this instance configuration, Yamcs fails to start and displays the following messages:
```shell
16:28:10.834 viper [1] YamcsServerInstance Loading service XtceTmRecorder
16:28:10.865 viper [1] YamcsServerInstance Cannot create service XtceTmRecorder, with arguments {}: null
16:28:10.866 _global [1] YamcsServer Instance viper failed: java.lang.NullPointerException
	org.yamcs.xtceproc.Subscription.addSequenceContainer(Subscription.java:52)
	org.yamcs.xtceproc.XtceTmExtractor.<init>(XtceTmExtractor.java:59)
	org.yamcs.xtceproc.XtceTmExtractor.<init>(XtceTmExtractor.java:42)
	org.yamcs.archive.XtceTmRecorder$StreamRecorder.<init>(XtceTmRecorder.java:195)
	org.yamcs.archive.XtceTmRecorder.createRecorder(XtceTmRecorder.java:138)
	org.yamcs.archive.XtceTmRecorder.init(XtceTmRecorder.java:113)
	org.yamcs.YamcsServer.createService(YamcsServer.java:777)
	org.yamcs.YamcsServer.createServices(YamcsServer.java:200)
	org.yamcs.YamcsServerInstance.doInit(YamcsServerInstance.java:131)
	org.yamcs.YamcsInstanceService.initAsync(YamcsInstanceService.java:62)
	org.yamcs.YamcsServerInstance.init(YamcsServerInstance.java:111)
	org.yamcs.YamcsServer.addInstance(YamcsServer.java:540)
	org.yamcs.YamcsServer.addGlobalServicesAndInstances(YamcsServer.java:1215)
	org.yamcs.YamcsServer.prepareStart(YamcsServer.java:1076)
	org.yamcs.YamcsServer.main(YamcsServer.java:880)

16:28:10.866 _global [1] YamcsServer Failure while attempting to validate configuration: java.lang.NullPointerException
	org.yamcs.xtceproc.Subscription.addSequenceContainer(Subscription.java:52)
	org.yamcs.xtceproc.XtceTmExtractor.<init>(XtceTmExtractor.java:59)
	org.yamcs.xtceproc.XtceTmExtractor.<init>(XtceTmExtractor.java:42)
	org.yamcs.archive.XtceTmRecorder$StreamRecorder.<init>(XtceTmRecorder.java:195)
	org.yamcs.archive.XtceTmRecorder.createRecorder(XtceTmRecorder.java:138)
	org.yamcs.archive.XtceTmRecorder.init(XtceTmRecorder.java:113)
	org.yamcs.YamcsServer.createService(YamcsServer.java:777)
	org.yamcs.YamcsServer.createServices(YamcsServer.java:200)
	org.yamcs.YamcsServerInstance.doInit(YamcsServerInstance.java:131)
	org.yamcs.YamcsInstanceService.initAsync(YamcsInstanceService.java:62)
	org.yamcs.YamcsServerInstance.init(YamcsServerInstance.java:111)
	org.yamcs.YamcsServer.addInstance(YamcsServer.java:540)
	org.yamcs.YamcsServer.addGlobalServicesAndInstances(YamcsServer.java:1215)
	org.yamcs.YamcsServer.prepareStart(YamcsServer.java:1076)
	org.yamcs.YamcsServer.main(YamcsServer.java:880)
```
Yamcs will start for me using this configuration, but only if the entire "subLoaders:" section is removed from the instance configuration.

If needed I can upload an example configuration that causes the problem for me.

xpromache, 2020-07-30:
Seems to come from a base container that does not have a restriction
criteria. Do you have something like that?

I think we should detect it and throw an error in the xtce parser.

merose, 2020-07-30:
Ah, that makes sense. The flight s/w team is still working on the XTCE and they have some containers that are missing <RestrictionCriteria> on the <BaseContainer>. Will try that, thanks!

Throwing an error for something Yamcs cannot handle would be helpful. However, it may be a limitation that could be overcome in Yamcs. In this case the container is probably configured to be abstract. That is, the derivation is probably something like this:

A (abstract) --> B (abstract) --> C (concrete)

In this case it would be convenient *not* to have a <RestrictionCriteria> on container B, since it's abstract. The <RestrictionCriteria> is not required by the XTCE schema, so it's unclear whether the XTCE designers wanted it to be present. A reasonable interpretation might be that restriction criteria is only strictly required on containers that will actually be matched against incoming packets.

leighgarbs, 2020-07-30:
In Yamcs 4.10.9 under some conditions we've seen a "inherits without defining an inheritance condition" error message on our configuration files, although for some reason (drowsiness perhaps) I can't get that message to appear now.

I can also see that the subloading functionality isn't the problem.  If I change the example configuration I originally posted to this ...
```yaml
mdb:

  - type: xtce
    args:
      file: xtce1.xml
      
  - type: xtce
    args:
      file: xtce2.xml
      
    subLoaders:
      - type: xtce
        args:
          file: xtce1.xml    # this is the same file as the first
```
... then Yamcs 5.0.0 will start and run.  As @merose said, once we get our XTCE changed so everything with a base container also has a restriction criteria, we should be past this.

xpromache, 2020-07-30:
> they have some containers that are missing on the . Will try that, thanks!
> In this case it would be convenient *not* to have a on container B, since
> it's abstract. The is not required by the XTCE schema, so it's unclear

I hadn't checked the XTCE but I see now that it is allowed so we should get
it to work inside Yamcs also.

commit 54534023 (2020-08-02): allow xtce tm container inheritance without restriction criteria. Fixes #428
closed 2020-07-31 by xpromache