yamcs/yamcs #432: Get Parameter Samples API does not use "count" parameter correctly

merose, 2020-08-24:
I have a parameter that has 18 values in a given 30 minute period. When using the `count` parameter of the Get Parameter Samples API I expect to get that many intervals returned. However, the number of intervals returned does not match. Am I misinterpreting that parameter?

In all uses of `curl` below, the `start` and `stop` times are the same.

When asking for 3 intervals, I get 1: (or when asking for 18, or 100 intervals, same behavior)

    $ curl 'http://localhost:8090/api/archive/myprojet/parameters/Demo/Mobility/roverPose.x/samples?start=2020-08-24T21:03:09.064Z&stop=2020-08-24T21:33:09.064Z&count=3'
    {
      "sample": [{
        "time": "2020-08-24T21:32:51.064Z",
        "avg": 0.0,
        "min": 0.0,
        "max": 0.0,
        "n": 18
      }]
    }
    
When I vary `count`, I get varying number of samples:

| count | samples |
| --- | --- |
| 1 | 1 |
| 3 | 1 |
| 18 | 1 |
| 100 | 1 |
| 1000 | 10 |
| 2000 | 15 |

I can verify that there are 18 values in the parameter archive for that 30 minute period as follows:

    $ curl -s 'http://localhost:8090/api/archive/myproject/parameters/Demo/Mobility/roverPose.x?start=2020-08-24T21:03:09.064Z&stop=2020-08-24T21:33:09.064Z&limit=100' | grep '"engValue"' | wc -l
          18

xpromache, 2020-08-25:
How are your values distributed in the 30 minutes period?
I think it divides that period in a number of equal duration intervals and
provides the data for each.
Some intervals may have no data that's why they are not sent.

merose, 2020-08-25:
Yes, of course you're right. Not a problem. Will close.

closed 2020-08-25 by merose

xpromache, 2020-08-25:
On this topic, we have now for a few months an intern looking to improve
the capabilities of Yamcs in "data analytics".
We hope to extend or create queries such as  the /samples or the /ranges to
do various things.
If you have some suggestions, please put them forward.

merose, 2020-08-25:
One suggestion for /samples: it is important for some clients to know the exact times for the minimum and maximum values in the range. This is currently a problem when zooming plots in OpenMCT, for example, with the way we are using /samples to get the data points to plot.