yamcs/yamcs #434: List Parameter Types API is not usable because type names are not returned

merose, 2020-08-24:
The List Parameter Types API returns an array of parameter types in the "types" attribute:

    GET /api/mdb/{instance}/parameter-types

However, the elements of that array do not contain any indication of the type name, so the data is not usable:

    {
      "types": [{
        "engType": "enumeration"
      }, {
        "engType": "enumeration"
      }, {
        "engType": "integer"
      }, {
        "engType": "time"
      }, {
        "engType": "aggregate",
        "member": [{
          "name": "Seconds",
          "type": {
            "engType": "integer"
          }
        }, {
          "name": "Subseconds",
          "type": {
            "engType": "integer"
          }
        }]
      },
      ...
    }

There is no separate API to list the parameter type names either. I don't think we need a separate API, but instead should add fields within each object in `types` that specifies the type name and the qualified name, as is done for the List Parameters API.

fqqb, 2020-09-07:
This API method was added as an incomplete experiment.
Would you have a use for it? Because otherwise, I'd rather remove than fix it.

MDB "Parameters" are already a sort of "type" definition, so I'm not yet fully convinced that Yamcs needs to expose the full XTCE-like type structure.

merose, 2020-09-08:
I do not have a use for this API method. I just saw that if I were to use it, it would be difficult because of the lack of names. It seems more likely that a client would use the List Parameters API method, which provides access to the type information.

fqqb, 2020-09-09:
ok. I've removed this method for the time being.
I think these named types are primarily useful when manually crafting these XML files as a way to limit the verbosity of type definitions that can be shared by several parameters.

closed 2020-09-09 by fqqb