yamcs/yamcs #436: Would be nice to have a wildcard option for the XTCE loader to load all files matching a pattern

merose, 2020-08-28:
For our mission we have over 20 subsystems, each with its own XTCE file. (There are also two "common" files that define base types, containers, and commands.) It would be convenient to have some way of asking the XTCE loader to load all files from a directory, to avoid such a long configuration in yamcs.<instance>.yaml, and also to avoid having to change that file for separate branches which may have different sets of XTCE (for testing, for example).

Something like this, for example:

    mdb:
      - type: xtce
        args:
          file: mdb/Base.xml
        subloaders:
          - type: xtce
            args:
              fileset: mdb/subsystems/*.xml

(One could get fancy and support an array of "fileset" values, of course, but I'm not sure that's necessary.)

xpromache, 2020-09-29:
This feature has been merged to the master, please give it a try if you can.
You can use both a glob pattern as well as a list of files or patterns:
```
...
  - type: xtce
    args:
       fileset: ["a*.xml", "b[1-3].xml"]

```
One trick is that if you have the fileset option matching multiple files (i.e. multiple SpaceSystems which are all at the same level in the hierarchy), you cannot anymore use subLoaders because we don't know to which of the loaded parent systems to attach the child (I could have chosen at random to attach it to the first one but since with those patterns it is not well specified which the first file matching the pattern is, so I think it is better to be explicit).

closed 2020-09-29 by xpromache

merose, 2021-02-04:
Forgot to comment on this previously. The wildcard patterns are really nice, thanks!