yamcs/yamcs #437: Need to indicate which argument is invalid if command does not pass validity checks

merose, 2020-08-28:
Currently (Yamcs 5.1.1), if you configure a <MetaCommand> which has <ValidRange> checks on its arguments, the Yamcs web UI will give you an error if you try to issue a command with arguments out of range. However, you just get a "Bad Request" error at the top of the page, with no indication of which argument is out of range. Further, there is no indication in the UI of what the valid range is.

To reproduce:

1. Configure a <MetaCommand> in XTCE with an argument whose type has a <ValidRange> attached.
2. In the Yamcs web UI, in Commanding, try to issue the command, specifying an argument value that is out of range.
3. Try to send the command.

Expected result: Error message indicates which argument is invalid.
Actual result: "Bad Request" error, but no indication of what is wrong.

merose, 2020-08-28:
Would also be good to show valid ranges in the commanding UI and in the MDB UI.

fqqb, 2020-08-28:
Quick answer to the first, the detailed error message should already work fine in Today's 5.1.2 release. It was originally working like that, then broke. I fixed it in the context of some other commit (https://github.com/yamcs/yamcs/commit/2d6052ad4f606579f2789ba05f89ae43f1e7642a)

I indeed agree that more indicators should be available like type-checking and valid ranges in the dialog.

merose, 2020-08-29:
Will close this since it's fixed in 5.1.2.

closed 2020-08-29 by merose