yamcs/yamcs #438: XtceStaxReader should not warn on <AncillaryDataSet> tag

merose, 2020-08-29:
We are using the XTCE definitions to generate C/C++ header files for use with cFS on a spacecraft. To add more data needed to the XTCE we are using `<AncillaryDataSet>` tags, allowed in the XTCE schema. For example:

    <AggregateParameterType name="HwSuspKin">
        <LongDescription>Susp Kin</LongDescription>
        <AncillaryDataSet>
            <AncillaryData name="STRUCT_ALIGNMENT">1</AncillaryData>
        </AncillaryDataSet>
        <MemberList>
            <Member typeRef="uint8" name="cntr"/>
            <Member typeRef="float32" name="relVel"/>
            <Member typeRef="float32" name="jointPos"/>
            <Member typeRef="float32" name="jointVel"/>
            ...
    </AggregateParameterType>   

(In this case, the `STRUCT_ALIGNMENT` value indicates a packed C/C++ structure.)

When we load this XTCE in Yamcs there are numerous warnings on startup about these tags:

    11:58:45.027 _global [1] XtceStaxReader Skipping unknown tag AncillaryDataSet at 127:36
    11:58:45.027 _global [1] XtceStaxReader Skipping unknown tag AncillaryData at 128:59
    11:58:45.039 _global [1] XtceStaxReader Skipping unknown tag AncillaryDataSet at 153:36
    11:58:45.039 _global [1] XtceStaxReader Skipping unknown tag AncillaryData at 154:59
    ...

Since this is valid XTCE, it would be nice not to have all of these warnings.

(In the future it would also be nice to make ancillary data available in the MDB API. Currently the only additional information that is returned to clients is `<AliasSet>`. We are using that tag for purposes beyond aliases, since we cannot use `<AncillaryDataSet>`.)

xpromache, 2020-09-01:
I pushed a commit which should get rid of the warnings.
MDB API still todo.

fqqb, 2020-09-09:
Addition of ancillary data to MDB API now also on master

closed 2020-09-09 by fqqb