yamcs/yamcs #439: There is no way to view all array values for <ArrayParameterType> parameters in the web UI

merose, 2020-09-02:
When viewing a parameter which has an `<ArrayParameterType>` type definition in the XTCE, there is no way to view all array values in the Yamcs web UI. For example, here is a view of an array parameter that has 32 array elements. As can be seen, only the first 5 array elements are shown, and there is no way to get details to show the rest of the array values.

[image]

merose, 2020-09-02:
There also appears to be no way to plot array values.

merose, 2020-09-02:
It *is* possible to view array values by manipulating the URL in the web UI. By adding "[21]" to the URL we can see the array value and also plot the value. The URL is now "http://localhost:8090/telemetry/parameters/%2FViperRover%2FNss%2Fdetector1PulseHeights%5B21%5D/summary?c=viper__realtime". However, the parameter type appears to be incorrect. It shows `integer[]`, which is the type for the entire array, rather than `integer`.

[image]

fqqb, 2021-04-20:
Closing as I think all of this has been addressed by now via various commits.

closed 2021-04-20 by fqqb