yamcs/yamcs #442: Need to get time of min and max values when using Get Parameter Samples api

merose, 2020-09-11:
In our client we using the Get Parameter Samples API, as does the Yamcs web UI, to display plots. We show the minimum and maximum value stacked vertically for each pixel horizontally, if there are more pixels than actual telemetry values during the plot time period. However, we would like to show the actual time of the minimum and maximum value if the user hovers over a plotted point. Currently the `time` field in the `Sample` structure is the midpoint of the sample interval.

It would be best if optional values were added to `Sample`:

- the time corresponding to the minimum value as a UTC string (no smearing around leap seconds)
- the time corresponding to the minimum value as `google.protobuf.Timestamp` (with smearing around leap seconds)
- the time corresponding to the maximum value as a UTC string (no smearing around leap seconds)
- the time corresponding to the maximum value as `google.protobuf.Timestamp` (with smearing around leap seconds)

Something like this:

    message Sample {
        optional string time = 1;
        optional double avg = 2;
        optional double min = 3;
        optional double max = 4;
        optional int32 n = 5;
        optional google.protobuf.Timestamp minTime = 6;
        optional string minTimeUTC = 7;
        optional google.protobuf.Timestamp maxTime = 8;
        optional string maxTimeUTC = 9;
    }

fqqb, 2022-08-10:
On master.

Added two protobuf timestamp fields `minTime` and `maxTime` which in JSON represent as strings.

Also started a refactoring of the existing `time` field from string to protobuf timestamp, this should have no effect on the JSON serialization.

The other points in this issue are also captured in #441, so closing.

fqqb, 2022-08-10:
See https://docs.yamcs.org/yamcs-http-api/parameter-archive/get-parameter-samples/

closed 2022-08-10 by fqqb