yamcs/yamcs #443: Commanding from YAMCS Studio 1.3.4 to YAMCS Server 5.1.3 severes connection

nmaas87, 2020-09-11:
I upgraded my existing YAMCS Server instance from 4.10.9 to 5.1.3. I have been using the YAMCS Studio in version 1.3.2 and 1.3.4 - now using 1.3.4. Everything works, but as soon as I send a TC, the connection between Studio and server dies / my OPI shows "disconnected" and I get no data anymore and have to manually reconnect to YAMCS server. On the server webinteface, I see that the TC command was received (Command History) - and I can manually reconnect studio to server without issue - but losing the connection on each commando is really problematic.

fqqb, 2020-09-11:
Studio 1.3.x was developed before Yamcs 5, so there may indeed be incompatibilities. If the connection drops, it’s likely some issue in the websocket callback logic. You should be able to find a stacktrace in ~/.config/yamcs-studio/logs

For Yamcs 5, you can for now try https://github.com/yamcs/yamcs-studio/releases/tag/v1.4.0.rc1 <https://github.com/yamcs/yamcs-studio/releases/tag/v1.4.0.rc1>
It has some other issues that I’m actively working to fix for a non-rc version (forecasted for next week).

> I upgraded my existing YAMCS Server instance from 4.10.9 to 5.1.3. I have been using the YAMCS Studio in version 1.3.2 and 1.3.4 - now using 1.3.4. Everything works, but as soon as I send a TC, the [...]

- Confidentiality Notice -
This communication and the information it contains is intended (a) for the person(s) or organisation(s) named above and for no other person or organisation, and (b) may contain confidential, legally privileged and protected content.
Unauthorized use, copying or disclosure is prohibited. If you received this communication in error, please notify the sender immediately, destroy any copies and delete it from your computer system.
Space Applications Services disclaims any and all liability if this email transmission was virus corrupted, altered or falsified.

nmaas87, 2020-09-11:
Thank you, I got no stacktrace, but I see the disconnect messages here:

````
ep 11 08:17:25.606 org.yamcs.studio.css.core.pvmanager.ParameterChannelHandler [43] [FINE] onStudioConnect called on /PAYLOAD/PAYLOAD/lo
Sep 11 08:17:25.607 org.yamcs.studio.css.core.pvmanager.ParameterChannelHandler [43] [FINE] PV connect on /PAYLOAD/PAYLOAD/lo
Sep 11 08:17:25.607 org.yamcs.studio.css.core.pvmanager.ParameterChannelHandler [44] [FINE] PV connect on /PAYLOAD/PAYLOAD/lo
Sep 11 08:17:30.326 org.yamcs.studio.core.YamcsPlugin [37] [FINE] Notify downstream components of Studio disconnect
Sep 11 08:17:30.326 org.yamcs.studio.core.YamcsPlugin [37] [FINE]  -> Inform ManagementCatalogue
Sep 11 08:17:30.327 org.yamcs.studio.core.YamcsPlugin [37] [FINE]  -> Inform TimeCatalogue
Sep 11 08:17:30.327 org.yamcs.studio.core.YamcsPlugin [37] [FINE]  -> Inform ParameterCatalogue
Sep 11 08:17:30.327 org.yamcs.studio.core.YamcsPlugin [37] [FINE]  -> Inform CommandingCatalogue
Sep 11 08:17:30.328 org.yamcs.studio.core.YamcsPlugin [37] [FINE]  -> Inform AlarmCatalogue
Sep 11 08:17:30.328 org.yamcs.studio.core.YamcsPlugin [37] [FINE]  -> Inform EventCatalogue
Sep 11 08:17:30.328 org.yamcs.studio.core.YamcsPlugin [37] [FINE]  -> Inform LinkCatalogue
Sep 11 08:17:30.510 org.yamcs.studio.core.YamcsPlugin [37] [FINE]  -> Inform ArchiveCatalogue
Sep 11 08:17:30.510 org.yamcs.studio.core.YamcsPlugin [37] [FINE]  -> Inform ConnectionUIHelper
Sep 11 08:17:30.510 org.yamcs.studio.core.YamcsPlugin [37] [FINE]  -> Inform PVCatalogue
Sep 11 08:17:30.515 org.yamcs.studio.core.YamcsPlugin [37] [FINE]  -> Inform AuthorizationStateProvider
Sep 11 08:17:30.516 org.yamcs.studio.core.YamcsPlugin [37] [FINE]  -> Inform ConnectionStateProvider
Sep 11 08:17:30.516 org.yamcs.studio.core.YamcsPlugin [37] [FINE]  -> Inform ConnectionStringStatusLineContributionItem
Sep 11 08:17:30.516 org.yamcs.studio.core.YamcsPlugin [37] [FINE]  -> Inform ProcessorStatusLineContributionItem
Sep 11 08:17:30.517 org.yamcs.studio.core.YamcsPlugin [37] [FINE]  -> Inform ConnectionStringStatusLineContributionItem
Sep 11 08:17:30.517 org.yamcs.studio.core.YamcsPlugin [37] [FINE]  -> Inform ProcessorStatusLineContributionItem
Sep 11 08:17:30.517 org.yamcs.studio.core.YamcsPlugin [37] [FINE]  -> Inform ParameterChannelHandler
Sep 11 08:17:30.517 org.yamcs.studio.css.core.pvmanager.ParameterChannelHandler [37] [FINE] PV disconnect on /PAYLOAD/PAYLOAD/ax
Sep 11 08:17:30.517 org.yamcs.studio.core.YamcsPlugin [37] [FINE]  -> Inform ParameterChannelHandler
````

But the rc1 version does work, thank you very much 👍🏻

closed 2020-09-11 by nmaas87