yamcs/yamcs #445: Subscribe Parameters API JSON return value does not match documentation

merose, 2020-09-14:
I have subscribed to a single parameter using the WebSocket API. I am getting responses, but they don't match the `SubscribeParametersData` specification at https://yamcs.org/docs/yamcs-http-api/processing/subscribe-parameters/. I'm getting the response below. As you can see, it does not contain the `mapping` attribute, so it's not possible to determine which value is for which parameter unless you keep track of the order in which you subscribed to parameters. This was not necessary in the old `_websocket` API.

[image]

merose, 2020-09-14:
I'm closing this for now, as it looks like I'm supposed to save the numericId-to-symbolicId mapping in the initial response. I was ignoring that response, in favor of using the `id` fields in the `ParameterValue` objects returned. If I find problems I'll report as issues, but it looks like I'm misunderstanding the API. (I think the documentation could be improved, though, with an example, too.)

closed 2020-09-14 by merose

fqqb, 2020-09-15:
It's an optimization to reduce the amount of strings that get sent. On the verbose JSON API the impact is negligible, but it does make a significant difference on the Protobuf API.

Some day the plan is to publish a reusable JavaScript library that does the dirty work of doing everything over Protobuf (like we have now for Python and Java).

Yes I'll prepare a few clarifications to the subscribe-parameters websocket page.